ARMmbed/mbed-os #8178: device-management init does not allow disabling certificate download

teetak01, 2018-09-19:
mbed dm init etc... always downloads a certificate. There should be a way to disable this.

0xc0170, 2018-09-19:
@teetak01 Can you leave the template in  place (Description, Issue type)  - they are useful to have

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-477*

aashishc1988, 2019-02-01:
@teetak01 , do you mind clarifying few things for me?
First, what you are referring to as "certificate" here, is "mbed_cloud_dev_credentials.c", correct? Or are you referring to ".update-certificates" folder that gets created?
Also, can you please let me know what the use case of this request is so that we address it in correct manner.

aashishc1988, 2019-02-12:
Ping!

teetak01, 2019-02-13:
Just ability to disable the download of mbed_cloud_dev_credentials.

I opened this mainly as consideration for CI-usage, where we would not need to download cerficates via mbed-tools, but use our pre-deployed credentials.

aashishc1988, 2019-02-13:
"mbed_cloud_dev_credentials" is not getting downloaded, it is already a part of the repo, I guess you meant that we need an option that will disable writing to that file with credentials, thus saving some time, correct?

40Grit, 2019-02-14:
I'll probably make a specific issue for it tomorrow but I am 💯 for anything that gets us closer to the annihilation of "mbed_cloud_dev_credentials.c

Please please 🙏

aashishc1988, 2019-02-14:
@teetak01 I talked to Brian and he mentioned that there was already a discussion going on a JIRA thread regarding adding an option  (something like --no-cert-download) that would prevent writing credentials to mbed_cloud_dev_credentials.c. Is that something we are looking for here?

teetak01, 2019-02-15:
@aashishc1988 something like that. Just having a option for would be nice.

aashishc1988, 2019-02-28:
We can mark this closed. It seems like I dont have permission to do this.

cmonr, 2019-02-28:
@aashishc1988 Feel free to ping @ARMmbed/mbed-os-maintainers in the future.

There are some things that GitHub's current PR permissions aren't granular enough for.

closed 2019-02-28 by cmonr