ARMmbed/mbed-os #8184: Stack overflow in tickless mode with debug profile on NUCLEO-L476RG

janjongboom, 2018-09-19:
### Description

Same as #7222? This is on Mbed OS 5.9 (949cb49ab0a144da0e3b04b6af46db0cd2a20d75) with mbed-os-example-lorawan on NUCLEO-L476RG:

```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x8015BF9
Error Value: 0x1
Current Thread: Id: 0x200020E8 Entry: 0x8015BAD StackSize: 0x200 StackMem: 0x20002178 SP: 0x10007EC0 
-- MbedOS Error Info --
```

Upping the idle thread stack size to 0x400 solves the issue.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-09-20:
cc @ARMmbed/team-st-mcd

bcostm, 2018-09-21:
This stack size is defined in `rtos/TARGET_CORTEX/mbed_rtx_conf.h`

/** The idle thread's stack size can be configured by the application, if not explicitly specified, it'll default to 512 */
#ifndef MBED_CONF_APP_IDLE_THREAD_STACK_SIZE
#define MBED_CONF_APP_IDLE_THREAD_STACK_SIZE    512
#endif

As this value can be changed by the application I don't know what can we do on our side ?

jeromecoutant, 2018-09-21:
Are you expecting something like:

#ifndef MBED_CONF_APP_IDLE_THREAD_STACK_SIZE
#ifdef MBED_TICKLESS
#define MBED_CONF_APP_IDLE_THREAD_STACK_SIZE 1024
#else
#define MBED_CONF_APP_IDLE_THREAD_STACK_SIZE 512
#endif
#endif

janjongboom, 2018-09-22:
@bcostm I know that it can be set by the application, but enabling tickless (a supported feature) mode on a NUCLEO board (a supported board) should never return in an instant hard fault. I thought this was acknowledged  and fixed in #7222.

jeromecoutant, 2018-10-01:
Hi

For information, I am currently testing TICKLESS feature with NUCLEO-L476RG.
Tests are OK in default mode, but often crash in Stack overflow as soon as I added some stats (MBED_CPU_STATS_ENABLED in my case)...

Maybe ARM core experts can double check ?

Thx

jeromecoutant, 2018-10-02:
@0xc0170 @deepikabhavnani

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6578*

jeromecoutant, 2018-10-08:
@c1728p9

janjongboom, 2018-10-24:
@bulislaw @0xc0170 ^

c1728p9, 2018-10-25:
I took a look at the stack overflow on debug builds with the NUCLEO-L476RG and can confirm that the background thread is overflowing. It looks like most of the extra stack usage is coming from the low power ticker wrapper which is only used on the background thread when tickless is enabled. Below is a breakdown of the stack usage.

Stack function | Stack pointer | Increase |   | Notes
-- | -- | -- | -- | --
update_present_time() | 0x200015e0 | 120 |   | Stack start = 0x20001070
schedule_interrupt() | 0x20001658 | 56 |   | Stack end = 0x20001870
ticker_insert_event_us() | 0x20001690 | 32 |   | Used = 656
mbed::TimerEvent::insert_absolute() | 0x200016b0 | 32 |   |  
mbed::Ticker::setup() | 0x200016d0 | 32 |   |  
mbed::Ticker::attach_us() | 0x200016f0 | 24 |   |  
LowPowerTickerWrapper::_schedule_match() | 0x20001708 | 96 |   |  
LowPowerTickerWrapper::set_interrupt() | 0x20001768 | 16 |   |  
lp_ticker_wrapper_set_interrupt() | 0x20001778 | 16 |   |  
schedule_interrupt() | 0x20001788 | 56 |   |  
ticker_insert_event_us() | 0x200017c0 | 32 |   |  
mbed::TimerEvent::insert_absolute() | 0x200017e0 | 32 |   |  
rtos::internal::SysTimer::schedule_tick() | 0x20001800 | 56 |   |  
rtos::internal::SysTimer::suspend() | 0x20001838 | 16 |   |  
default_idle_hook() | 0x20001848 | 16 |   |  
rtos_idle_loop() | 0x20001858 | 8 |   |  
osRtxIdleThread() | 0x20001860 | 16 |   |  
osThreadGetId() | 0x20001870 |   |   |

jeromecoutant, 2018-10-26:
@janjongboom Could you review #8551 ?
Thx