ARMmbed/mbed-os #8185: atmel-rf-driver: Components get build as default?

deepikabhavnani, 2018-09-19:
### Description

Building project mbed-os_development (ARM_MUSCA_A1, ARMC6)
Scan: mbed-os_development
Scan: env
Compile [  0.2%]: mbed_tz_context.c
Compile [  0.3%]: NanostackRfPhyAtmel.cpp
Compile [  0.5%]: at24mac.cpp
[ERROR] In file included from .\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\at24mac.cpp:16:
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source/at24mac.h:71:11: error: no type named 'I2C' in namespace 'mbed'
    mbed::I2C _i2c;
    ~~~~~~^
1 error generated.

1. Aren't component suppose to be in Component_XXX folder, and not always enabled as default?
2. Error is seen when DEVICE_I2C is not present, atmel-rf-driver should be enabled only when DEVICE_I2C is present.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug  

Reported by @gaborkertesz

0xc0170, 2018-09-19:
@SeppoTakalo Please review

theotherjimmy, 2018-09-27:
It looks like the whole driver should be guarded with `#if defined(DEVICE_I2C)` or similar.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-348*

deepikabhavnani, 2018-10-10:
Fixed by https://github.com/ARMmbed/mbed-os/pull/8327

closed 2018-10-10 by deepikabhavnani