ARMmbed/mbed-os #8188: NRF52 interrupts already activated by fall/rise call

ChiSaw, 2018-09-19:
### Description
    In `InterruptIn::fall(..)` and `InterruptIn::rise(..)` calling `gpio_irq_set` in target https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/gpio_api.c#L221 , the interrupts get activated per default. This results in direct ISR calls on hardware if the interrupt input pin is floating during boot up for example. Per default interrupts shouldn't be activated after hooking up an ISR with fall or rise, right? Only after explicitly calling `InterruptIn::enable_irq()`.

    - Target: NRF52832
    - Toolchain: (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git
    - mbed-cli 1.8.1 / mbed-os 5.9
    - Steps to reproduce:
        - empty mbed project
        - main.cpp
          ```
          #include "mbed.h"

         volatile uint8_t count = 0;
         void handler() {
             count++;
         }

          void main() {
              InterruptIn interrupt(p0);
              interrupt.fall(callback(&handler));
              while(1);
          }
```
and then toggling p0 on hw e.g. calls handler() for me

### Issue request type
    [ ] Question
    [ ] Enhancement 
    [X] Bug

0xc0170, 2018-09-20:
As the example in the InterruptIn header file shows, fall/raise method enabled interrupts - only required to have a callback. Activated by default.

enable/disable irq were added later, to be able to disable/re-enable only interrupts for specific input (in most cases it is actually the entire port).

drahnr, 2018-09-20:
From an API POV this does not make much sense, since commonly one wants to attach a callback once, and the enable disable based on some business logic.
Also the API documentation says nothing about the irq being activated https://os.mbed.com/docs/v5.9/mbed-os-api-doxy/classmbed_1_1_interrupt_in.html#a774eed83bdbd9a95ea530b2eee9b7754
If it is expected to read the source code every single time, then generated documentation becomes pointless.

0xc0170, 2018-09-20:
> From an API POV this goes against a common pattern - one wants to attach a callback once, and the enable disable based on some business logic.

This is how `attach` functionality is designed in the drivers (you can review `attach` methods, the object does not have enable/disable these callbacks). Either valid callback or NULL (former enables, latter disables).

> Also the API documentation says nothing about the irq being activated 

Please review @ARMmbed/mbed-os-hal

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-349*

adbridge, 2020-10-13:
@drahnr We've updated our automation to use a new issue template, I will fix the requirements .

closed 2021-10-20 by ciarmcom