ARMmbed/mbed-os #8191: STM32F103: set_time doesnt set the time correctly

guest253456, 2018-09-20:
### Description

```
int main()
{
        wait_ms(2000);

        printf("Startup: %ld\r\n", time(0));
        set_time(1537429507);
        printf("After setup: %ld\r\n", time(0));
        printf("Error count: %u\r\n", mbed_get_error_count());

        wait_ms(100);

        NVIC_SystemReset();
}
```

Running the above code i get the following output

> Startup: 27909
> After setup: 1537429507
> Error count: 0
> Startup: 27909
> After setup: 1537429507
> Error count: 0
> Startup: 27909
> After setup: 1537429507
> Error count: 0
> Startup: 27909
> After setup: 1537429507
> Error count: 0
> Startup: 27909

I'm using Mbed 5.9.7(949cb49).

 

[ ] Question  
[ ] Enhancement  
[x ] Bug

bcostm, 2018-09-20:
I checked in our ci tests history and all RTC tests are PASS on this board using mbed-os 5.9.7. I also rechecked all RTC tests on the master branch and all is PASS.

The RTC Reset is tested correctly but is this test correct or maybe does not cover this use case ?

```
+-------------------+---------------+------------------------------+--------+--------------------+-------------+
| target            | platform_name | test suite                   | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+------------------------------+--------+--------------------+-------------+
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | OK     | 30.62              | default     |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | OK     | 50.68              | default     |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc_reset     | OK     | 21.05              | default     |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc_time      | OK     | 17.41              | default     |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc_time_conv | OK     | 35.75              | default     |
+-------------------+---------------+------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 5 OK
mbedgt: test case report:
+-------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite                   | test case                                                       | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Functional Test: RTC counts seconds.                            | 1      | 0      | OK     | 10.06              |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_0.                      | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_1.                      | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Functional Test: set time - CUSTOM_TIME_2.                      | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: attach original RTC functions.                       | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: attach stub RTC functions.                           | 1      | 0      | OK     | 0.07               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: set_time() - RTC functions are defined.              | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: set_time() - init RTC function is undefined.         | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: set_time() - write RTC function is undefined.        | 1      | 0      | OK     | 0.1                |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: time() - RTC functions are defined, RTC is disabled. | 1      | 0      | OK     | 0.1                |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: time() - RTC functions are defined, RTC is enabled.  | 1      | 0      | OK     | 0.09               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: time() - isenabled RTC function is undefined.        | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: time() - read RTC function is undefined.             | 1      | 0      | OK     | 0.08               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_drivers-rtc       | Unit Test: time() - result is stored in given buffer.           | 1      | 0      | OK     | 0.07               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | RTC - accuracy                                                  | 1      | 0      | OK     | 10.04              |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | RTC - enabled                                                   | 1      | 0      | OK     | 0.03               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | RTC - glitch                                                    | 1      | 0      | OK     | 4.04               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | RTC - init                                                      | 1      | 0      | OK     | 0.05               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc           | RTC - persist                                                   | 1      | 0      | OK     | 4.04               |
[... cut ...]
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years full support     | 1      | 0      | OK     | 9.95               |
| NUCLEO_F103RB-ARM | NUCLEO_F103RB | tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years partial support  | 1      | 0      | OK     | 10.21              |
+-------------------+---------------+------------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
```

guest253456, 2018-09-21:
>The RTC Reset is tested correctly
Which test is that? Does it perform a system reset and check the value? As you can see from my output, `set_time` works but loses it after a reboot.

jeromecoutant, 2018-09-21:
Hi
Issue is confirmed.

I quickly checked the driver file from ST cube:
https://github.com/ARMmbed/mbed-os/blob/8f7c0ae18d5fbe171b1ff61a52f1152a0957cc43/targets/TARGET_STM/TARGET_STM32F1/device/stm32f1xx_hal_rtc.c#L57

I am expecting some feedback from ST RTC driver expert.

Regards,

bcostm, 2018-09-21:
> Which test is that? Does it perform a system reset and check the value?

The test is here https://github.com/ARMmbed/mbed-os/tree/master/TESTS/mbed_hal/rtc_reset

This was my initial remark. This test should FAIL on F1 devices... today it is not the case.

jeromecoutant, 2018-09-21:
Hi @guest253456 

Is it possible for you to check #8213 ?

@bcostm 
Why rtc_reset test has always been OK ?
Because set time in the tests is inferior than 24h (1 day)...

guest253456, 2018-09-21:
> Is it possible for you to check #8213 ?

Yes, its working with that patch.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6581*

PR #8213 (2018-10-10): STM32F1 RTC : save values in register
### Description

RTC drivers for STM32F1 family is quite different than other families.
Date feature is currently managed only by SW.

With this patch, RTC read and write has been quite simplified,
and values are now saved into register.

Fixes #8191 

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-10 by cmonr