ARMmbed/mbed-os #8194: Requirements: mbed-cloud-sdk having exact version match

0xc0170, 2018-09-20:
### Description

`mbed-cloud-sdk==2.0.0`  - I would like to understand this requirement. We have a new PR that is updating this to 2.0.1. Are we expecting to have exact match and update for specific patch releases that we choose?

Why not to loose this dependency to accept at least patch releases ? How often is patch release expected vs req update in this repository ?

@ARMmbed/mbed-os-tools @MarceloSalazar 

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

cmonr, 2018-09-20:
Related to: https://github.com/ARMmbed/mbed-os/pull/8160

My opinions: https://github.com/ARMmbed/mbed-os/pull/8160#issuecomment-422462165

0xc0170, 2018-09-20:
@arekzaluski Please review

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTTOO-3457*

0xc0170, 2018-11-27:
> @arekzaluski Please review

any update for this issue?

theotherjimmy, 2018-11-27:
@0xc0170 If anything, I used to be using internal APIs. https://github.com/ARMmbed/mbed-os/blob/master/tools/device_management.py#L94-L134 is the code affected and no longer uses internal APIs. We should be good to relax that requirement now.

0xc0170, 2018-11-28:
Good to hear ! Please send a fix and we can close this one.

0xc0170, 2019-08-26:
Not relevant anymore.

closed 2019-08-26 by 0xc0170