ARMmbed/mbed-os #8196: UBLOX_EVK_ODIN_W2 target failed on mbed HAL lp_ticker/sleep/rtc greentea test

jamesbeyond, 2018-09-20:
### Description
`UBLOX_EVK_ODIN_W2` target sharing the same HAL implentation with STM targets, where STM targets passing all the greentea test. but `UBLOX_EVK_ODIN_W2` filed on following mbed greentea tests:
* tests-mbed_hal-lp_ticker
* tests-mbed_hal-common_tickers
* tests-mbed_hal-rtc
* tests-mbed_hal-sleep

### Toolchians:
GCC_ARM, ARM, IAR

### mbed version:
latest master branch  SHA `0233c8df4`

### Steps to reproduce
```
mbed test  -m UBLOX_EVK_ODIN_W2 -t GCC_ARM -n <above-greentea-test>
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-09-20:
@ARMmbed/team-ublox Please review

ashok-rao, 2018-09-20:
@RobMeades @andreaslarssonublox  - can you please check this? Thanks.

screamerbg, 2018-09-21:
@asifrizwanubx @andreaslarssonublox 

Notice that the ODIN W262 is working fine. The failures only happen on ODIN W260 (external antenna variant)

asifrizwanubx, 2018-09-24:
> Notice that the ODIN W262 is working fine. The failures only happen on ODIN W260 (external antenna variant)

We will soon look into it.

asifrizwanubx, 2018-09-24:
TE_OM2-237

jamesbeyond, 2018-09-24:
Hi, @asifrizwanubx
Just to update the latest investigation from our side. both W262 and W260 seem to have the problem the with tests. The decsrip[tion of the issue remains accurate.

asifrizwanubx, 2018-09-25:
@screamerbg @jamesbeyond 

It is not a hardware issue. I am also facing same failure here with the old batch of HW. In PR [6913](https://github.com/ARMmbed/mbed-os/pull/6913) they are passing but in PR [7579](https://github.com/ARMmbed/mbed-os/pull/7579) they are not and even without Access Point API commit.

It seems me some change b/w these PR's effect the test cases. I am investigating the issue, could you please confirm me the old names of the above test case?. IINW these test cases were renamed.

jamesbeyond, 2018-09-25:
Hi @asifrizwanubx,
The failed tests are been added by the PR #7009, which got merged between #6913 and #7579.

in PR #7009, we merged 3 feature branches of the new HAL API `us/lp_ticker`, `sleep` and `rtc` design, along with HW drivers updates from each venders. This is the point when ST updated their underneath drivers for all STM family targets. Those new tests got intriduced at same time.

asifrizwanubx, 2018-09-25:
@jamesbeyond if they are new tests it's mean there is no previous result of them?

jamesbeyond, 2018-09-25:
@asifrizwanubx You are right, there is no previous result for those exact tests cases

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6580*

aqib-ublox, 2020-03-11:
@adbridge above mentioned tests are passing please see below solution:

1. UBLOX_EVK is having J21 jumper exposed to connect LSE oscillator explicitly.
2. If J21 is not connected , update mbed_app.json to mark lse_available=false
its mention on platform page as well, please see https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/ 

see below logs:
[Logs.txt](https://github.com/ARMmbed/mbed-os/files/4317388/Logs.txt)

jamesbeyond, 2020-03-11:
Thanks @aqib-ublox, your solution been tested, and confirmed working.
close this issue

closed 2020-03-11 by jamesbeyond