ARMmbed/mbed-os #8197: low DAC output voltage nucleo-l432kc

kubazdz, 2018-09-20:
### Description
Hi, I'm new to mbed and im trying to use nucleo l432kc DAC as an analog output. I'm using very simple code, and the output voltage is very low (330mV) and I'm wondering why. Here's the code:

```
#include "mbed.h"

AnalogOut tri(PA_4);

int main() {
    tri = 1;
}
```
in my opinion it should be 3.3V... Is it some hardware issue?

### Issue request type

[X] Question  
[  ] Enhancement  
[  ] Bug

kubazdz, 2018-09-24:
Yes, the same on PA_5. Well it seems that something is wrong with my board, thanks

bcostm, 2018-09-24:
You can try also to configure the PA_4 and PA_5 pins as `DigitalOut` to see if you can see the 3.3V. But I'm pretty sure (unfortunately) that your board is damaged...

kubazdz, 2018-09-25:
But when i do sawtooth or sine wave, it works fine, but the range is from 0 to 330mV

kubazdz, 2018-09-25:
Ok, took a different board and works fine, thanks

bcostm, 2018-09-25:
Great. Can you please close this issue now ? Thanks.

closed 2018-09-25 by 0xc0170