ARMmbed/mbed-os #8198: Compiler warning on BlockDevice

BramdeBoer1, 2018-09-20:
### Description

Target:      STM32L072
IDE:           IAR 8.30.1
Mbed OS: 5.10.0 RC3

After upgrading the os from 5.9.3 to 5.10.0 i get a lot of the following compiler warnings.
Warning[Pe068]: integer conversion resulted in a change of sign 

It is due to BD_ERROR_DEVICE_ERROR being used as return value when the return type is of type bd_size_t.
BD_ERROR_DEVICE_ERROR is defined as -4001 whilst bd_size_t is unsigned.

The related file is BlockDevice.h

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

davidsaada, 2018-09-20:
@BramdeBoer1 An example of such an error, including compiled file (C/CPP file) and line number would be very helpful.

BramdeBoer1, 2018-09-20:
An example of the issue can be found in BufferedBlockDevice.cpp:254

```
bd_size_t BufferedBlockDevice::get_erase_size() const
{
    if (!_is_initialized) {
        return BD_ERROR_DEVICE_ERROR;
    }

    return _bd->get_erase_size();
}
```

[image]

davidsaada, 2018-09-20:
@BramdeBoer1 Thanks.
@yossi2le Please handle.

yossi2le, 2018-09-26:
I have created PR #8256 to fix this issue.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTSTOR-659*

closed 2019-06-05 by ciarmcom