ARMmbed/mbed-os #8199: Can't initialize NB-IoT SIM with Ublox SARA-N211

henningray, 2018-09-20:
### Description

Hello, I'm having trouble initializing a NB-IoT Sim card with my Ublox SARA-N211 (Firmware 6.57) via an OnboardCellularInterface. I can power up the modem successfully but in PPPCellularInterface::initialize_sim_card() I keep getting AT Error messages:

> AT> AT+CPIN?
> AT? +CPIN: %*15[^
> ]
> %n
> AT< 
> AT! ERROR
> AT(Aborted)
 
Also, after the device tries to reconnect I receive 
> AT< +UFOTAS: 0,1

even though I am not sending a UFOTAS command.
I am using Mbed OS 5.9.0 with GCC_ARM toolchain.

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCELL-1419*

tho068, 2018-10-05:
Having the exact same issue. Just bought the C030-N211 devkit, and it is not working with my Telenor nbiot sim card. I suspect I need a firmware update for the nbiot modem, but this is nowhere to be found.

jarvte, 2018-10-11:
Hi, Sorry for the delay!

PPPCellularInterface and OnboardCellularInterface are deprecated API's.
You should use cellular framework: https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.9/features/cellular

There is also mbed-os version 5.10 already available.

You can check from https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9/features/cellular/easy_cellular/EasyCellularConnection.cpp how Cellular API's are used to connect or use EasyCellularConnection class for easy connection like in example application: https://github.com/ARMmbed/mbed-os-example-cellular

Supported modules are in https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets

Br,
Teppo

henningray, 2018-10-11:
The main problem for me (and possibly for @tho068 ) was that the OnboardCellularInterface (and also EasyCellularConnection) uses several AT commands that are not supported on the SARA N2XX like the above AT+CPIN or AT+CGREG. So you either have to manually remove or replace them in the code or use the UbloxATCellularInterfaceN2xx: https://os.mbed.com/teams/ublox/code/example-ublox-cellular-interface/

tho068, 2018-10-11:
Yeah. That is true, so I ended up creating my own very small lib which worked perfectly.

—
Med vennlig hilsen

Thomas Holden
MSc, computer science

jarvte, 2018-10-12:
New cellular implementation folder https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.10/features/cellular/framework/AT implements 3GPP AT commands. If target (SARA N2XX) differs from those then it needs a target to folder https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.10/features/cellular/framework/targets where module/modem specific commands can be overridden.

You also need to specify target in mbed_app.json if it's not already in https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/features/cellular/framework/common/CellularTargets.h

closed 2018-10-23 by ciarmcom