ARMmbed/mbed-os #8200: CM3DS 'EMAC unicast long' Timeout

kapi90, 2018-09-20:
When I run EMAC related Greentea tests on my CM3DS target (MPS2+ Rev. C) it seems that the 600s Timeout that is set by default is not enough for the EMAC unicast long testcase, in which the board sends, receives and validates 50,000 packets. When I ran the same test without using the python environment (Connect to it with DS-5 and Serial Terminal (Putty), do a software reset and send {{__sync;1}} to start) it didn't have the 600s timeout limit and so all test cases passed with no problem. 

Target: CM3DS
Toolchain: GCC_ARM 7 2017-q4-major
Mbed OS SHA: 62f8b922b420626514fd4690107aff4188469833

[ ] Question  
[x ] Enhancement  
[ ] Bug  

@ashok-rao @SeppoTakalo

ashok-rao, 2018-09-27:
@SeppoTakalo ..what do you think for the above issue?

SeppoTakalo, 2018-09-27:
How long did it then take on this platform?

@mikaleppanen What do you think, is 600 seconds not enough?

SeppoTakalo, 2018-09-27:
I'm okay with the idea of rising the timeout if its not way too long.
We are generally not expecting any defined speed or performance from the devices.

So if somebody sends a PR with test report that shows that this device was able to pass, I'm happy to accept.

mikaleppanen, 2018-09-27:
For K64F ethernet it takes 84.17 seconds. I'm also ok with raising the timeout if the device ethernet speed is such that 600 seconds is not enough.

SeppoTakalo, 2018-09-27:
@kapi90 Can you share the Ethernet driver with us?
Perhaps send a PR.

Within the 5.9 timeframe all ethernet drivers have been refactored and outdated dropped. By that time, nobody was working with the CM3DS, so it got dropped.

If you have ported the driver to new EMAC API we are happy to review and test it.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3908*

kapi90, 2018-10-09:
Hi @SeppoTakalo, sorry for the late reply, I just got back from vacation. I'll need to push my code through our internal code and IP reviews first, once that is done I'm gonna share the EMAC driver, so you can take a look at it and test it however you like. How does that sound ?

kapi90, 2018-10-16:
There's been a PR created for the EMAC driver, that's available here: https://github.com/ARMmbed/mbed-os/pull/8444

The issue seems to have changed, now the timeout happens at the end of the last test case, look like 1200s is a little bit short to get to the end of all test cases.

TuomoHautamaki, 2018-11-06:
Fix provided in https://github.com/ARMmbed/mbed-os/pull/8604

closed 2018-11-07 by ciarmcom