ARMmbed/mbed-os #8201: When Will 'feature-watchdog' be Released?

mattbrown015, 2018-09-20:
### Description
The product I'm working on needs a watchdog.  Precise requirements are vague.

After searching for 'watchdog' I found the [feature-watchdog](https://github.com/ARMmbed/mbed-os/tree/feature-watchdog) branch and some PRs and issues relating to it.  My initial reaction was "ah, great!".

What I didn't find was an overview of the status.  Is there a way to find out how a feature branch is progressing and when it might be merged into master?  Or, can anyone say when 'feature-watchdog' might be merged into master.

Given that I need to make some progress on a watchdog in the next week or 2 I'm guessing that 'feature-watchdog' won't be merged soon enough for me.  So, is there a way that I can clone master and then pull a feature branch into my clone?

I working with an STM32L4 and, AFAICT, 'feature-watchdog' supports this device.

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-09-21:
@scartmell-arm Please review

scartmell-arm, 2018-09-21:
The watchdog feature was in the feature-branch to give partners time to add support for watchdog on their platforms before merging into master. There should be no harm in merging it now but it will most likely be available in the next release 5.11.

If the platform you are working on already has an implementation for the API there should be no harm in pulling in the feature branch in your own repository to use it. There are no changes planned for the API so it will be stable.

mattbrown015, 2018-09-21:
That answered my question, thanks!

5.11 will be too late for me and I'm not yet sure whether to use STM32L4 IWDG or WWDG and how to 'tickle the dog'.  But what's in 'feature-watchdog' is good background.

closed 2018-09-21 by mattbrown015

daniel-cesarini-tridonic-com, 2018-09-24:
How will the functionalities of IDWG and WWDG be discriminated in the current Watchdog API?