ARMmbed/mbed-os #8202: Icetea: TypeError: a bytes-like object is required, not 'str'

MarceloSalazar, 2018-09-20:
### Description

During OOB for 5.10 release, found issues while evaluating icetea:

- Mbed CLI 1.8.2
- Mbed OS rc3
- OSX El Capitan
- Python 3.7

```
git clone --branch mbed-os-5.10.0-rc3 https://github.com/ARMmbed/mbed-os 5.10-rc3
cd 5.10-rc3/mbed-os
mbed test -m K64F -t GCC_ARM --compile --icetea -vv
...
[mbed-85384] Exec "/usr/local/opt/python/bin/python3.7 -u /Users/marsal01/mbed/OOB/5.10-rc3-2/mbed-os/tools/run_icetea.py -m K64F -t GCC_ARM -v --application-list" in "/Users/marsal01/mbed/OOB/5.10-rc3-2/mbed-os"
Traceback (most recent call last):
  File "/Users/marsal01/Library/Python/3.7/lib/python/site-packages/mbed/mbed.py", line 3338, in main
    status = pargs.command(pargs)
  File "/Users/marsal01/Library/Python/3.7/lib/python/site-packages/mbed/mbed.py", line 1985, in thunk
    return command(**argv)
  File "/Users/marsal01/Library/Python/3.7/lib/python/site-packages/mbed/mbed.py", line 2878, in test_
    if applications_to_add and 'TEST_APPS-' in applications_to_add:
TypeError: a bytes-like object is required, not 'str'
[mbed-85384] ERROR: Unknown Error: a bytes-like object is required, not 'str'
---
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

jamesbeyond, 2018-09-20:
I can reproduce on Ubuntu 16.04 with Python 3.7. Both with IceTea and GreenTea.
IceTea throws the error right away.
GreenTea throws the error in the end.

But the IceTea error is from [mbed-cli `mbed.py`](https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2878) and GreenTea error is from [mbed-os `build_api.py`](https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L1360)

0xc0170, 2018-09-21:
@OPpuolitaival

OPpuolitaival, 2018-09-24:
Fix: https://github.com/ARMmbed/mbed-os/pull/8181

0xc0170, 2018-09-24:
Fixed via 8181

closed 2018-09-24 by 0xc0170