ARMmbed/mbed-os #8203: Unable to drive RGB LED on WIZWIKI W7500 and W7500P's through DigitalOut Interface

naveenkaje, 2018-09-20:
### Description
WIZWIKI W7500 and W7500P's RGB LED D1 does not turn on as expected with a simple program.
This issue was raised by a user on [questions](https://os.mbed.com/questions/82435/Blink-example-does-not-work-on-WIZwiki-W/)
```
#include "mbed.h"

DigitalOut myled(LED1);
int main() {
     myled = 0;
      while(1) {  
            wait_ms(1000);
	    myled = !myled;
     }
}
```
[WIZWIKI W7500 Target Page](https://os.mbed.com/platforms/WIZwiki-W7500/)
[Schematic](http://wizwiki.net/wiki/lib/exe/fetch.php?media=products:wizwiki_w7500:wizwiki_w7500_v1.1.pdf)

### Issue request type

[ ] Question  
[ ] Enhancement  
[x ] Bug

naveenkaje, 2018-09-20:
Adding @andrewc-arm

khj098765, 2018-09-21:
@andrewc-arm @naveenkaje 
mbed online compiler mbed-os Library version is release 163 (mbed library. Release version 163)
I tried it with mbed online compiler. But the operation was strange.
I tried it locally with ARM, IAR, GCC_ARM. The operation was normal.
Why is this happening despite using the same Library(rev 163)?

0xc0170, 2018-09-21:
the online compiler uses ARMCC (default profile) plus also prebuilt binary , therefore might be something there related.

How did you test locally @khj098765  ?

khj098765, 2018-09-27:
@0xc0170 
Online is the latest version mbed lib and local is mbed-os Rev163.
online : https://os.mbed.com/users/mbed_official/code/mbed/
local : https://github.com/ARMmbed/mbed-os/releases/tag/mbed_lib_rev163

Both online and local compiled with ARMCC.
[Local compile log file](https://github.com/ARMmbed/mbed-os/files/2422127/log.txt)
[online&local compiled bin files](https://github.com/ARMmbed/mbed-os/files/2422125/mbedissue.zip)

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1651*

naveenkaje, 2018-10-04:
@khj098765 @andrewc-arm 
Based on my experiments today, builds based on 510 CLI using ARMCC work as expected. But the Online compiler built binary for the same code does not work.

khj098765, 2018-10-04:
@naveenkaje 
Yes! That's what I want to say.
It is very strange.
What should I do to solve this problem?

naveenkaje, 2018-10-05:
The issue seems to be particular to the online compiler. When a project is created on CLI from scratch, it compiles and works as expected. But on the online compiler built version does not work.

I created the project on online compile and then imported it to the CLI. It doesn't really seem to build. It reports successful build in 14 seconds which is suspicious.

```
mbed import https://os.mbed.com/users/naveenkaje/code/mbed_blinky_w7500/

$ time mbed compile -m WIZWIKI_W7500 -t ARM -c
Using targets from C:\mbed-cli\mbed_blinky_w7500\mbed\e95d10626187\targets.json
Building project mbed_blinky_w7500 (WIZWIKI_W7500, ARM)
Scan: mbed_blinky_w7500
Compile [100.0%]: main.cpp
[Warning] main.cpp@63,0: #1-D: last line of file ends without a newline
[Warning] main.cpp@53,0: #177-D: variable "status" was declared but never referenced
Link: mbed_blinky_w7500
Elf2Bin: mbed_blinky_w7500

Module	.text	.data	.bss
---------------------	---------------	-----------	-------------
BUILD\WIZWIKI_W7500	851(+851)	0(+0)	168(+168)
[lib]\c_p.l	10791(+10791)	16(+16)	280(+280)
[lib]\cpprt_p.l	78(+78)	0(+0)	0(+0)
[lib]\fz_ps.l	2294(+2294)	0(+0)	0(+0)
[lib]\m_ps.l	44(+44)	0(+0)	0(+0)
[lib]\mbed.ar	4812(+4812)	34(+34)	168(+168)
anon$$obj.o	32(+32)	0(+0)	0(+0)
mbed\e95d10626187	6783(+6783)	112(+112)	641(+641)
Subtotals	25685(+25685)	162(+162)	1257(+1257)
Total Static RAM memory (data + bss): 1419(+1419) bytes
Total Flash memory (text + data): 25847(+25847) bytes

Image: .\BUILD\WIZWIKI_W7500\ARM\mbed_blinky_w7500.bin

real 0m14.688s
user 0m0.000s
sys 0m0.078s
```

theotherjimmy, 2018-10-05:
@naveenkaje That's an mbed 2 project, so of course it builds quickly.

naveenkaje, 2018-10-05:
@theotherjimmy thanks for pointing that out.

I think the issue is that the default templates that is offered on online compiler for Wizwiki are all mbed os 2 based. See attached screenshort.

[image]

@khj098765 : Do you need to use mbed os 2 or can you be based off of mbed-os 5.x? 

It seems like by default, the online compiler is picking mbed 2 for your project. If you move to mbed-os 5, the program works as expected.

To use mbed os 5, create a blinky program for a board like K64F and then change the target to choose Wizwiki W7500. I know it is not the ideal way to create a program, but we have some limitation w.r.t project templates for WizWiki that we are trying to fix. I have tried this method and the blinky program works as expected on Wizwiki W7500 with mbed-os.5 based build.

Do you have any requirements on which OS you need to have?

Here is a document about mbed 2 and mbed os 5. https://os.mbed.com/docs/latest/introduction/mbed-os-2-and-mbed-os-5.html

MarceloSalazar, 2020-04-14:
Closing as target won't be supported in Mbed OS 6 - https://github.com/ARMmbed/mbed-os/pull/12775

closed 2020-04-14 by MarceloSalazar