ARMmbed/mbed-os #8205: Sleep tracer is not finding matching driver when performing an unlock

TacoGrandeTX, 2018-09-20:
### Description

We are not properly performing the driver lookup when we perform an unlock of the deep sleep counter.  This only affects the stats counter and not the sleep counter itself, but it leads to confusing results when using the sleep tracer to monitor sleep.

Test Program:

```
#include "mbed.h"

Timer timmy;

int main() {

    // starting a timer increments sleep lock count
    timmy.start();

     // wait will bump the sleep count and decrement it
    wait (3);

    // stop the timer... this should reduce the sleep lock count
    timmy.stop();

}  // end main()
```
This results in:
```
LOCK: Timer.cpp, ln: 54, lock count: 1
LOCK: mbed_wait_api_rtos.cpp, ln: 45, lock count: 2
Sleep locks held:
[id: Timer.cpp, count: 1]
[id: mbed_wait_api_rtos.cpp, count: 1]
UNLOCK: mbed_wait_api_rtos.cpp, ln: 47, lock count: 1
Unlocking sleep for driver that was not previously locked: Timer.cpp, ln: 68
Sleep locks held:
[id: Timer.cpp, count: 1]
```
It still shows the timer holding a sleep lock when it has actually been released.

If we revert commit 9a0e8797b9b9fc32f6ce8f5fc508a43ff76b42f8 then the problem goes away.  We were doing a strcmp instead of a pointer compare in sleep_tracker_find():

https://github.com/ARMmbed/mbed-os/commit/9a0e8797b9b9fc32f6ce8f5fc508a43ff76b42f8#diff-4ef16dda46f5130a57b304fed2459ee3L53

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

TacoGrandeTX, 2018-09-20:
@scartmell-arm Could you please comment?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-489*

cmonr, 2019-02-12:
@scartmell-arm Ping?

TacoGrandeTX, 2019-02-12:
@cmonr We have a dialogue on the internal JIRA and I'm now testing the PR

closed 2019-02-14 by 0xc0170