ARMmbed/mbed-os #8207: Icetea: [Warning] Test with name 'test_cmdline' was not found in the available tests

MarceloSalazar, 2018-09-20:
### Description

Testing Mbed OS 5.10-RC3

While trying to run Icetea and the test_cmdline example, I was stuck trying to understand how to select this particular tests, using the option `-n`.

Every time I ran this line:
```
mbed test -m K64F -t ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED
```

Saw this:
```
[Warning] Test with name 'test_cmdline' was not found in the available tests
```

Which made me think I was doing something wrong, and trying many other things...

Thanks to @jamesbeyond I found the test actually compiles and works.

This has to be fixed to help the user understand what's going on:
- If the test cannot be found, the build should stop immediately
- If the test can be found, the warning should not be shown

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-09-21:
cc @OPpuolitaival

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBEDOSTEST-203*

closed 2018-11-01 by ciarmcom