ARMmbed/mbed-os #8209: Custom target in separate directories

vidavidorra, 2018-09-21:
### Description
In the documentation [Adding custom targets](https://os.mbed.com/docs/latest/tools/adding-and-configuring-targets.html) I see that a file `custom_targets.json` can be used in a separate directory from the mbed-os targets if that directory is given to the `mbed` commands using the `--source` flag. We normally do local builds using the GCC ARM compiler and an `mbed export` for CMake. Beside the JSON file, is it also possible to add custom target folders there?
Say I want to add a the STM32F205RG as custom target. Can I add a folder for that (e.g. `TARGET_STM32F205RG`) in a the same directory as where I can create the `custom_targets.json` and add all my files for that specific targets (all in a separate directory than the mbed-os targets).

E.g. following dir structure and provide `--source custom_targets` to `mbed export` where I use `-m STM32F205RG` to select my custom target.

```
.
|-- mbed-os
|-- custom_targets
|   |-- custom_targets.json
|   |-- TARGET_STM
|   |   |-- TARGET_STM32F2
|   |   |   |-- TARGET_STM32F205RG
```

As far as I have read the docs, I only saw mention of the custom JSON file but not of the separate directory with targets.

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

vidavidorra, 2018-09-24:
I've tested it myself and found that it indeed was possible with the `cmake_gcc_arm` exporter.

I've used a structure as follows. Note that the microcontroller manufacturer and/or family files that are already in Mbed OS (`mbed-os/targets/TARGET_STM` and `mbed-os/targets/TARGET_STM/TARGET_STM32F2`) will be used, so the custom target only needs to contain the device specific files.
```
./custom_mbed_os_targests
|-- TARGET_STM
|   `-- TARGET_STM32F2
|       `-- TARGET_STM32F205RG
|           |-- device
|           |   |-- TOOLCHAIN_GCC_ARM
|           |   |   |-- STM32F205ZGTx_FLASH.ld
|           |   |   `-- startup_stm32f205xx.S
|           |   |-- cmsis.h
|           |   |-- cmsis_nvic.h
|           |   |-- flash_data.h
|           |   |-- hal_tick.h
|           |   |-- memory_zones.h
|           |   |-- stm32f205xx.h
|           |   |-- stm32f2xx.h
|           |   |-- system_clock.c
|           |   `-- system_stm32f2xx.h
|           |-- low_level_cellular_modem_api.c
|           |-- low_level_cellular_modem_api.h
|           |-- onboard_modem_api.c
|           |-- PeripheralNames.h
|           |-- PeripheralPins.c
|           `-- PinNames.h
|-- custom_targets.h
`-- custom_targets.json
```

As this is not documented I'll change the title for documentation, as I've tested that it works.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-496*