ARMmbed/mbed-os #8211: Losing error on constructor open() call

lauri-piikivi, 2018-09-21:
https://github.com/ARMmbed/mbed-os/blob/949cb49ab0a144da0e3b04b6af46db0cd2a20d75/features/netsocket/UDPSocket.h#L50

@kjbracey-arm there might be error here. The constructor hides the open calls from underneath, and thus different implementations may fail -- see for example open on 
https://github.com/ARMmbed/esp8266-driver/blob/master/ESP8266Interface.cpp

Note that this may impact cloud client, as they do not explicitly call open, only the constructor and connect

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3928*

SeppoTakalo, 2018-11-20:
Fixed in:
* https://github.com/ARMmbed/mbed-os/pull/8794
* https://github.com/ARMmbed/mbed-os-5-docs/pull/838

closed 2018-11-28 by ciarmcom