ARMmbed/mbed-os #8214: Improve rtc_reset test ?

jeromecoutant, 2018-09-21:
### Description

As seen in #8191, value set in the rtc_reset test has missed some issue...

https://github.com/ARMmbed/mbed-os/blob/082ba151ac39262e42775f066581ef0f787160bb/TESTS/host_tests/rtc_reset.py#L32

Do you think we should update this 50000 value to a bigger one ?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

jeromecoutant, 2018-09-21:
@guest253456

0xc0170, 2018-09-21:
cc @mprse

guest253456, 2018-09-24:
>Do you think we should update this 50000 value to a bigger one ?

It breaks if i add one more zero.

mprse, 2018-09-24:
Thanks @jeromecoutant.

What value should be used and why start value has influence on persistence of RTC after reset?

jeromecoutant, 2018-09-24:
Check #8191, we found an issue whereas RTC_TEST was OK.
So this issue is just a proposition to enhance test efficiency.
Value should be today's date... (more realistic value than 50000) ?

bcostm, 2018-09-24:
https://www.epochconverter.com/

Today is 1537778889

mprse, 2018-09-24:
I think it is worth to change the start value. I'll create the patch.

mprse, 2018-09-24:
Patch can be found here: PR https://github.com/ARMmbed/mbed-os/pull/8230.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-352*