ARMmbed/mbed-os #8215: drivers-lp_ticker test failed on NUCLEO_F746ZG

maciejbocianski, 2018-09-21:
### Description

- target: NUCLEO_F746ZG
- test: tests-mbed_drivers-lp_ticker
- toolchain: GCC_ARM (likely all)
- Mbed OS: master (082ba151a (HEAD -> master, upstream/master, origin/master, origin/HEAD) Merge pull request #8112 from studavekar/update_doc_unittest)
- catched in PR: #7925

```
[1537279728.87][CONN][RXD] >>> Running case #11: 'Test multi ticker'... 
[1537279728.97][CONN][INF] found KV pair in stream: {{__testcase_start;Test multi ticker}}, queued... 
[1537279729.17][CONN][RXD] :83::FAIL: Expected 16 Was 32 
[1537279729.27][CONN][INF] found KV pair in stream: {{__testcase_finish;Test multi ticker;0;1}}, queued... 
[1537279729.37][CONN][RXD] >>> 'Test multi ticker': 0 passed, 1 failed with reason 'Assertion Failed'
```

### How to reproduce
To catch the problem, test should be run multiple time or even modified to run test case constantly in loop

run `mbed test -t GCC_ARM -m NUCLEO_F746ZG -n tests-mbed_drivers-lp_ticker -v --profile develop`

@MateuszMaz with help of @mprse will continue investigation on this issue

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-09-21:
cc @ARMmbed/team-st-mcd

MateuszMaz, 2018-09-24:
I can not reproduce this error on current master(9d78650fccaf7debaf81d2010f30985de0ea0974) (even changed this test to perform it 20000 times). I could reproduce it when qspi_hal_test_refactoring was based on 91354184dc0282749d27cb9f6529e2a1811b7b8c (as a branch on @maciejbocianski  fork)

jeromecoutant, 2018-09-27:
Hi
I also tried to reproduce the issue (master branch), but I couldn't... :-(

maciejbocianski, 2018-10-01:
I also tested today on latest master on all compilers and cannot reproduce.

closed 2018-10-01 by maciejbocianski