ARMmbed/mbed-os #8217: BBC Microbit fails to build

fduignan, 2018-09-21:
### Description
Using the cloud based compiler on mbed.com  Mbed Revision 170:e95d106  fails to compile even basic blinky for BBC Microbit with the following error:
Error: Undefined symbol __heap_base (referred from /opt/armcc5_06_u3/bin/../lib/armlib/mc_p.l(malloc.o)).
Info: Unable to download. Fix the reported errors...

The compiler reports a version number of 1.10.18.0
The earlier mbed version : 169:a7c7b63 works fine

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-09-21:
Looking at the recent changes to the target directory, I suspect this is caused by recent changes to the boot process. Here's the file that is requesting that symbol https://github.com/ARMmbed/mbed-os/blob/88eb8aab2330c9ab2b8948079f123ba4b777224a/rtos/TARGET_CORTEX/TOOLCHAIN_ARM_MICRO/mbed_boot_arm_micro.c

I can't find the __heap_base in the startup file for `MCU_NRF51_16K_BASE` targets

@c1728p9 Please review

fduignan, 2018-09-27:
As of today, the online mbed for the bbc microbit appears to be completely borked.  Compiling the basic blinky results in:
Error: The build system did not finish successfully.  Error number -230.
Previously working programs (presumably with older library versions) compile successfully but appear to crash during boot.
I have not yet managed to successfully select an older library and compile a project.
Help?

fduignan, 2018-09-27:
More information: I'm suspecting an incorrect linker script in the online mbed compiler setup.
If I look at the (Intel) Hex file output from the compiler for a new program I seen entries like this:
:020000040001F9
:108000000040002099810100AB810100AD81010099
If I look at the hex output for a file I compiled a week or so ago I see this:
:020000040000FA
:10000000C0070000D1060000D1000000B1060000CA
This indicates that the linker is outputting code for a different address range today (8000) compared to a week or so ago (0000).  
Can someone verify and fix please?

0xc0170, 2018-10-01:
The latest report is another issue (linker issue?). Looking at the history of files for microbit (https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NORDIC/TARGET_MCU_NRF51822) no change for some time. Neither in the target definition (targets.json) file. 

I would suspect a change in tools rather

0xc0170, 2018-10-01:
@c1728p9 Can you review `__heap_base` heap base error?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-351*

PR #8305 (2018-10-10): Tools: Fix Microbit releases
### Description

I noticed that there was quite a bit missing from the mbed2 release of 
the microbit today. That's because the microbit uses ARM by default and
the GCC_ARM small library. The release script was trying to release for 
ARM using scan rules for uARM. Turns out we're stuck with 2 configuration
parameters for the same thing: `default_lib` for GCC and `default_toolchain`
for ARM. Dang

Resolves #8217 I think.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-10 by cmonr