ARMmbed/mbed-os #8220: STM32F405RG CCM (RAM) usage in mbed-os

vidavidorra, 2018-09-21:
### Description
What needs to be configured in order to use the full RAM capability of the STM32F405RG (or any MCU in that family). According to the [datasheet §2.2.6](https://www.st.com/resource/en/datasheet/stm32f405rg.pdf#page=21) it has 192 KB RAM total, of which 64 KB is CCM (core coupled memory). However, when I run the [mbed-memory-status](https://github.com/nuket/mbed-memory-status) code to print the heap, I see that it shows about the same RAM as my other 128 KB RAM MCU (STM32F205RG) has while I'd expect 192 KB. See snippet below for memory output.

```
	stack ( start: 2000A4B0 end: 2000BCB0 size: 00001800 used: 00001800 ) thread ( id: 2000BD04 entry: 08081461 name: main_thread )
	stack ( start: 20005148 end: 20005248 size: 00000100 used: 00000100 ) thread ( id: 20004FFC entry: 0808160D name: unknown )
	stack ( start: 20005048 end: 20005148 size: 00000100 used: 00000100 ) thread ( id: 20004FB4 entry: 08084D9D name: unknown )
	heap ( start: 2000BFBC end: 2001FC00 size: 00013C44 used: 000009F5 )	alloc ( ok: 00000002	fail: 00000000 )
isr_stack ( start: 2001FC00 end: 20020000 size: 00000400 used: 00000400 ) 
```

I'm not sure whether anything additional needs to be configured for the CCM to be used by mbed-os and whether the mbed-memory-status would be able to detect that.

In `targets/TARGET_STM/mbed_rtx.h` I see that `INITIAL_SP` is set to `0x20020000UL` which would be 128 KB dec. Doesn't this need to be `0x20030000UL` for 192 KB, or is it because the other 64 KB is CCM?

This is on mbed-os-5.9.0 and using offline build with cmake exporter and GCC toolchain (`7-2018-q2-update`).

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

LMESTM, 2018-09-21:
@vidavidorra the 2 RAMs are not contiguous. 
CCRAM base address is 0x10000000 so you cannot do what you propose. 
You may want to look at the linker script, for instance here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_MTS_MDOT_F405RG/device/TOOLCHAIN_IAR/stm32f405xx.icf
Te CCRAM may be used to store specific application buffers so you could specify in the linker script some application that you'd like to move to CCRAM. Alternatively, you could move the  either the stack of the heap to CCRAM, like:
```
place in RAM_region   { readwrite,
                        block HEAP };
place in CCMRAM_region   { block CSTACK };
```
But the choice is very application dependent, and each toolchain has its own format and capabilities ...

LMESTM, 2018-09-24:
@vidavidorra 
If this is GCC toolchain, you'd have to find examples out there, like:
https://mcuoneclipse.com/2012/11/01/defining-variables-at-absolute-addresses-with-gcc/
I think this should be something like:
unsigned char __attribute__((section (".CCM"))) stack_mem_user_app[APP_STACK_SIZE];
but need to be tested and checked with your application and environment.

vidavidorra, 2018-09-26:
@LMESTM I see, thanks for the reference. I'll try to test it this week, hopefully. Will post back here with results!

vidavidorra, 2018-09-28:
@screamerbg Yes, I'll do that. Most likely I have no time for it today, so next week.

vidavidorra, 2018-10-01:
@LMESTM I've tested this today and got it to compile successfully. Flashing however doesn't succeed. I've tried placint the `.myBufBlock` section as first thing in my sections and as last as well, but both with the same result. I can also see with `arm-none-eabi-nm` that the linker puts my memory (`test_mem`) in the correct address.

See the snippet below for the error I get when I try to flash the device. When I remove the added section and the creation and use of `test_mem` in the main I can flash it again. I'm using [stlink](https://github.com/texane/stlink) and an ST-LINK/V2 to flash my device.

```shell
st-flash --reset --format ihex write ./cmake-build-release/example-app.hex
st-flash 1.5.1
2018-10-01T15:58:03 INFO common.c: Loading device parameters....
2018-10-01T15:58:03 INFO common.c: Device connected is: F4 device, id 0x10076413
2018-10-01T15:58:03 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2018-10-01T15:58:03 INFO common.c: Attempting to write 134087680 (0x7fe0400) bytes to stm32 address: 134348800 (0x8020000)
2018-10-01T15:58:03 ERROR common.c: addr too high
stlink_fwrite_flash() == -1
```

### STM32F405.ld
```assembly
MEMORY
{
    FLASH (rx) : ORIGIN = MBED_APP_START, LENGTH = MBED_APP_SIZE
    CCM (rwx) : ORIGIN = 0x10000000, LENGTH = 64K
    RAM (rwx) : ORIGIN = 0x20000188, LENGTH = 128K - 0x188
}

SECTIONS
{
    /* placing my named section at given address: */
    .myBufBlock 0x10000000 :
    {
        KEEP(*(.myBufSection)) /* keep my variable even if not referenced */
    } > CCM

    /* Other sections. */
}
```

### main.cpp
```c++
std::uint8_t __attribute__((section (".myBufSection"))) test_mem[1024];

int main() {
    test_mem[1000] = 0xaa;

    printf("test_mem[1000]: %02hhx", test_mem[1000]);
}
```

closed 2018-10-01 by adbridge

vidavidorra, 2018-10-02:
Not sure why this was closed already, but I've got everything working now.
Turns out that I needed to mark the section as `NOLOAD` to ensure that its uninitialised and not part of the output files of the linker. After that I could flash it as normal.

vidavidorra, 2018-10-02:
For reference and anyone that has the same problem here is a small document I wrote during my research and for future reference.

# Define sections in linker file

The linker needs to know what memory areas are present in the device and where those are. Additionally, specific memory sections can be defined to be used in the application code.

 

## Memory mapping

The memory mapping used in this document is that of the [STMicroelectronics STM32F405RG](https://www.st.com/resource/en/datasheet/stm32f405rg.pdf#page=71). The linker file has the following memory mapping to define the three areas of the memory.

1. `FLASH`, flash memory starting at `0x08000000` with a size of 1 MiB.
2. `CCM`, core coupled memory (data RAM) starting at `0x10000000` with a size of 64 KiB.
3. `RAM`, SRAM starting at `0x20000000` with a size of 128 KiB.

```assembly
MEMORY
{
    FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 1024K
    CCM (rwx) : ORIGIN = 0x10000000, LENGTH = 64K
    RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 128K
}
```

## Sections

In order to use a named section in the application code, it needs to be defined in the linker file. The snippet below from the linker file defines a section in the CCM at start address `0x10000000`. The application code can use this named section to create a buffer for example. Below the snippet, some parts of this section are explained in more detail.

```assembly
SECTIONS
{
    /**
     * CCM (core coupled memory) data RAM section.
     * The section is marked as NOLOAD so it's not loaded at runtime. This
     * ensures that the section does not affect the flash size.
     */
    .ccm_data_ram 0x10000000 (NOLOAD) :
    {
        /** Align on 4 byte boundary. */
        . = ALIGN(4);
        
        /** Keep the variable even if it is not referenced. */
        KEEP(*(.ccm_data_ram))
    } > CCM
}
```

- `NOLOAD`

  > The section should be marked as not loadable, so that it will not be loaded into memory when the program is run. - [GNU ld - Output Section Type](https://sourceware.org/binutils/docs/ld/Output-Section-Type.html#Output-Section-Type)

  This is required to ensure that the section does not affect size of the output file of the linker. Because of this, the section is **NOT** initialised and contains random values when the program is started. [`std::memset`](https://en.cppreference.com/w/cpp/string/byte/memset) could be used at the start of the main to initialise the section to zero or any value for that matter.

- `ALIGN`

  > You can increase an output section’s alignment by using `ALIGN`. As an alternative you can enforce that the difference between the VMA and LMA remains intact throughout this output section with the `ALIGN_WITH_INPUT` attribute. - [GNU ld - Forced Output Alignment](https://sourceware.org/binutils/docs/ld/Forced-Output-Alignment.html#Forced-Output-Alignment)

- `KEEP`

  > When link-time garbage collection is in use (`--gc-sections`), it is often useful to mark sections that should not be eliminated. This is accomplished by surrounding an input section’s  wildcard entry with `KEEP()`, as in `KEEP(*(.init))` or `KEEP(SORT_BY_NAME(*)(.ctors))`. - [GNU ld - Input Section Keep](https://sourceware.org/binutils/docs/ld/Input-Section-Keep.html#Input-Section-Keep)

# Place a variable on a section

The `__attribute__` keyword can be used to place the variable in a specific section, `ccm_data_ram` in this case. The following snippet shows how a buffer named `ccm_buffer` is placed in the `ccm_data_ram` section and is used. Note that `ccm_buffer` is uninitialised in this example.

```c++
#include <cstdint>

#define CCM_BUFFER_SIZE (64 * 1024)
std::uint8_t __attribute__((section(".ccm_data_ram"))) ccm_buffer[CCM_BUFFER_SIZE];

int main() {
    printf("ccm_buffer at start:\n");
    for (std::uint32_t i = 0; i < CCM_BUFFER_SIZE; i++) {
        printf("%02hhx", ccm_buffer[i]);
        if ((i + 1) % 64 == 0) {
            printf("\n");
        }
    }
    
    /** Initialse to 0xaa. */
    memset(ccm_buffer, 0xaa, CCM_BUFFER_SIZE);
    
    printf("ccm_buffer after initialisation:\n");
    for (std::uint32_t i = 0; i < CCM_BUFFER_SIZE; i++) {
        printf("%02hhx", ccm_buffer[i]);
        if ((i + 1) % 64 == 0) {
            printf("\n");
        }
    }
}
```

> Normally, the compiler places the objects it generates in sections like `data` and `bss`. Sometimes, however, you need additional sections, or you need certain particular variables to appear in special sections, for example to map to special hardware. The `section` attribute specifies that a variable (or function) lives in a particular section.
>
> Use the `section` attribute with *global* variables and not *local* variables, as shown in the example. - [GNU GCC manual - Common Variable Attributes](https://gcc.gnu.org/onlinedocs/gcc-8.2.0/gcc/Common-Variable-Attributes.html#Common-Variable-Attributes)

# Verify the allocation

As part of the compilation, the linker will produce a `.elf` file. This file can be inspected by the [`nm` tool](https://sourceware.org/binutils/docs/binutils/nm.html#nm) where the created variable can be searched. The following snippet shows how this can be done for `arm-none-eabi-nm`. This shows where the variable is located in memory and in this case successfully shows that it is in the CCM section, i.e. at address `0x10000000`.

```shell
$ arm-none-eabi-nm -ln cmake-build-release/example.elf | grep 'ccm_buffer'
10000000 B ccm_buffer	main.cpp:4
```

jeromecoutant, 2018-10-02:
Hi
Fell free to push your new target in targets/TARGET_STM/TARGET_STM32F4/
Thx

vidavidorra, 2018-10-02:
@jeromecoutant I'm currently on mbed-os 5.9.0 and added my targets a fork we maintain there. I want to update to mbed-os 5.10.0 and than define the targets there. I've also written a document on how to create custom STM32 targets. I'll create a new ticket for that later because I'm missing information on how to create some specific files for the targets. Once I create that ticket I'll mention you!

After that I'll definitely create a PR for the targets. Those are the following two for now.
- STM32F205RG
- STM32F405RG