ARMmbed/mbed-os #8227: Ublox EVK ODIN W2 does not pass WIFI-CONNECT-PARAMS-CHANNEL-FAIL test case

SeppoTakalo, 2018-09-24:
### Description
Ublox EVK ODIN W2 does not pass WIFI-CONNECT-PARAMS-CHANNEL-FAIL test case.

```
[1537758884.87][CONN][RXD] >>> Running case #10: 'WIFI-CONNECT-PARAMS-CHANNEL-FAIL'...
[1537758884.87][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-CONNECT-PARAMS-CHANNEL-FAIL}}, queued...
[1537758887.16][CONN][RXD] :38::FAIL: Expression Evaluated To FALSE
[1537758887.26][CONN][INF] found KV pair in stream: {{__testcase_finish;WIFI-CONNECT-PARAMS-CHANNEL-FAIL;0;1}}, queued...
[1537758887.36][CONN][RXD] >>> 'WIFI-CONNECT-PARAMS-CHANNEL-FAIL': 0 passed, 1 failed with reason 'Test Cases Failed'
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

asifrizwanubx, 2018-09-24:
@0xc0170 

https://github.com/ARMmbed/mbed-os/pull/7579#issuecomment-407725014

It seems me that it was due to that missed change.

asifrizwanubx, 2018-09-24:
We removed that particular change, is it possible to view it from history? I havn't any local history of it. Otherwise we need some time.

asifrizwanubx, 2018-09-24:
I got it.

asifrizwanubx, 2018-09-24:
TE_OM2-238

0xc0170, 2018-09-24:
> It seems me that it was due to that missed change.

@asifrizwanubx  the comment you referenced as I recall was about the new API added in the PR (not only updating the drivers to fix issues - as fix) ? How can this be resolved?

asifrizwanubx, 2018-09-24:
It was related to Test Cases fix. We will soon create a new PR for test cases fix.

AmmadRehmat, 2018-09-25:
@SeppoTakalo  I have fixed the other issue https://github.com/ARMmbed/mbed-os/issues/8228#issuecomment-423918541 but this one I am unable to reproduce. This one is passing for me every time I run this test. I am attaching my mbed_app.json file, could you please share yours so I can try with those settings. 
[mbed_app.txt](https://github.com/ARMmbed/mbed-os/files/2414144/mbed_app.txt)

SeppoTakalo, 2018-09-25:
Then I need to re-check our test jobs.

This is the Json we use
[mbed_app.txt](https://github.com/ARMmbed/mbed-os/files/2414462/mbed_app.txt)

AmmadRehmat, 2018-09-25:
@SeppoTakalo  your json file shows that your unsecure and secure AP are running on same channel. For this test to pass you need to run them on different channels.

SeppoTakalo, 2018-09-25:
AH... You are correct
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/network/wifi/wifi_connect_params_channel_fail.cpp#L37

Test case has been written assuming that channel numbers are not the same.

I need to fix this.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3927*

PR #8241 (2018-10-08): Fix WIFI_CONNECT_PARAMS_CHANNEL_FAIL testcase.
### Description

Test case was assuming that secure and unsecure SSID were on different
channels.
This is not a requirement and it should be OK to run on same channel.

Fixed the testcase by using +1 on channel number to get a wrong channel.

Fixes #8227

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-08 by cmonr