ARMmbed/mbed-os #8228: Ublox EVK ODIN W2 does not pass WIFI-GET-RSSI test case

SeppoTakalo, 2018-09-24:
### Description
Ublox EVK ODIN W2 does not pass WIFI-GET-RSSI test case.

```
[1537758839.50][CONN][RXD] >>> Running case #5: 'WIFI-GET-RSSI'...
[1537758839.50][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-GET-RSSI}}, queued...
[1537758859.68][CONN][RXD] :36::FAIL: Expected 0 Was -3004
[1537758859.78][CONN][RXD] >>> 'WIFI-GET-RSSI': 0 passed, 1 failed with reason 'Test Cases Failed'
```

Same device is still able to pass unsecure and secure connectivity tests.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

asifrizwanubx, 2018-09-24:
Same problem.

> @0xc0170
> 
> #7579 (comment)
> 
> It seems me that it was due to that missed change.

asifrizwanubx, 2018-09-24:
TE_OM2-238

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6677*

screamerbg, 2019-02-11:
@aqib-ublox Could you please take a look at this?

aqib-ublox, 2019-02-11:
@screamerbg sure

AmmadRehmat, 2019-02-12:
@screamerbg  I think these issues were resolved in the referenced pull requests. Kindly check the referenced pull requests.

SeppoTakalo, 2019-02-12:
Ublox EVK ODIN W2 already passes this test case.

Closing this, as not valid anymore.

closed 2019-02-12 by SeppoTakalo