ARMmbed/mbed-os #8232: Target specific .mbedignore

borysekm, 2018-09-24:
### Description
I don't understand how to do some target specific .mbedignore. I'm using mbed-os-5.9 downloaded to PC and building it locally from the command line. I have custom targets defined in custom_targets.json, they are called BYZANCE_IODAG3E and BYZANCE_IODAG3E_BOOTLOADER.

I need to exclude all unnecessary folders during BYZANCE_IODAG3E_BOOTLOADER build. Those folders are:

```
events/*
features/*
components/*
rtos/*
```

**My current solution**
Before building BYZANCE_IODAG3E_BOOTLOADER I always create .mbedignore file in mbed folder with content specified above and build this target with a command like
`mbed/tools/build.py -m BYZANCE_IODAG3E_BOOTLOADER -t GCC_ARM --source mbed --build mbed-os_BYZANCE_IODAG3E_BOOTLOADER --no-archive --profile small`
Then I delete or rename this .mbedignore file (with some prefix) and build another target.
It works well and do exactly what I want, but this process is really annoying and clumsy.

**What I've tried**
'build.py' --help gave me some hint for possible solution
`--ignore IGNORE    Comma-separated list of patterns to add to mbedignore (eg. ./main.cpp)`
which could probably solve my problem - I just specify some ignore patterns during *_BOOTLOADER target build. But typing command 
`mbed/tools/build.py -m BYZANCE_IODAG3E_BOOTLOADER -t GCC_ARM --source mbed --build mbed-os_BYZANCE_IODAG3E_BOOTLOADER --no-archive --profile small --ignore mbed/rtos/*, mbed/events/*, mbed/features/*, mbed/components/*` gives me response like

`build.py: error: unrecognized arguments: mbed/events/*, mbed/features/*, mbed/components/802.15.4_RF mbed/components/storage mbed/components/wifi`

What I'm doing wrong? Or is there some other solution, how to exclude these folders during specific target build? Maybe using some custom_targets.json label? I have tried to disable all these features in corresponding mbed_lib.json files by setting "present" to 0, but it doesn't work (looks like this setting is ignored).

### Issue request type

[X ] Question  
[ ] Enhancement  
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-494*

theotherjimmy, 2018-10-23:
> I don't understand how to do some target specific .mbedignore.

Sorry for the delay in getting back to you. This is currently not possible.

closed 2018-11-13 by cmonr