ARMmbed/mbed-os #8240: Is errno Thread safe?

maskedw, 2018-09-25:
[X] Question  
[ ] Enhancement  
[ ] Bug  

I think errno needs Thread local storage space, can not define a Thread local variable in CMSIS-RTOS?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-492*

deepikabhavnani, 2018-11-27:
errno is not thread safe, and yes thread local storage is not supported in Mbed OS.
We have thread safe mechanism to capture error codes and status, and it slowly getting adopted in parts of Mbed OS.  Please see https://os.mbed.com/docs/v5.10/apis/error-handling.html for more details

SenRamakri, 2018-12-04:
Hello @maskedw, We think the above response from @deepikabhavnani answered your question. We are closing this issue for now, feel free to re-open if you have further queries on this.

closed 2018-12-04 by SenRamakri