ARMmbed/mbed-os #8264: Nucleo_STM32L476RG: Extra power consumption was detected when choosing HSI as LPUART clock source

Eman869, 2018-09-27:
### Description

I had measured the power consumption in deepsleep mode on Nucleo_STM32L476. With LPUART clock source overrided with HSI clock, it is around 200uA and it is too high. I found that HSI would be enabled in STOP mode, which causes the extra power consumption. High speed clock should be disabled during deepsleep mode. Here is the mbed_app.json which override the LPUART clock source.

mbed_app.json 
```
{
        "target_overrides": {
                "NUCLEO_L476RG": {
                        "target.lpuart_clock_source": "USE_LPUART_CLK_HSI",
                        "target.clock_source": "USE_PLL_HSE_EXTC"
                }
        }
}

```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-10-01:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2018-10-01:
Should be fixed by #8264

0xc0170, 2018-10-01:
> Should be fixed by #8264

#8284 ?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6680*

PR #8284 (2018-10-15): STM32: Only LSE clock should be enabled in STOP mode
Check LPUART clock source before enable it in STOP mode, only LSE could be enabled in STOP mode.

### Description

It is to fix the extra power consumption in STOP mode. HSI clock would cause extra power consumption if HSI clock is enabled in STOP mode. Only LSE clock could be enabled in STOP mode.

Fix #8264

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-15 by cmonr