ARMmbed/mbed-os #8265: Transfering data using SPI and SPI Slave issue

kubazdz, 2018-09-27:
### Description

Hi, I'm trying to make a simple program in mbed using two nucleos l432kc as a SPI master and slave. I want to send two floats from master to slave, multiply it by slave, and return result. Everything works fine except for returning result and after lot of debugging I can't find where the issue is. 

Master:

```
#include "mbed.h"
#include <stdio.h>

/* Initialize inputs and outputs */
SPI spi (PB_5, PB_4, PB_3);             // mosi, miso, sclk
DigitalOut cs (PA_4);
Serial pc(USBTX, USBRX); 

int transfer_float(union float_to_transfer transfer);

union float_to_transfer {
    float value;
    unsigned char bytes[4];
};

/* Initialize SPI and variables */
void spi_init() {
    cs = 1;
    spi.format(8,0);                   
    spi.frequency(20000000);
    return;
}

/* Send data to slave */
int send(union float_to_transfer transfer) {
    pc.printf("Sending value: %f\n\r", transfer.value); 
    spi.write(0);
    int i = 0;
    for(i = 0; i < 4; i++) {
        cs = 0;
        pc.printf("Sending: Byte %i is %02x\n\r", i, transfer.bytes[i]);
        spi.write(transfer.bytes[i]); 
        cs = 1;
    }    
    return 0;
}

/* Request data transfer from slave */
float receive() {
    union float_to_transfer transfer;
    spi.write(1);
    pc.printf("Requested transfer...\n\r");
    for(int i = 0; i < 4; i++) {
        cs = 0;
        transfer.bytes[i] = spi.write(0x02);        // dummy
        cs = 1;
    }
    for(int j = 0; j < 4; j++) {
        pc.printf("Received: byte %i is %02x\n\r", j, transfer.bytes[j]); 
    }
    return transfer.value;
}

int main() {
    
    spi_init();
    
    /* Declare unions */
    union float_to_transfer first_value;
    union float_to_transfer second_value;
    pc.printf("**** MASTER *****\n\r");
    while(1) {
        /*Read values*/        
        pc.printf("First number: \n\r");
        pc.scanf("%f", &first_value.value);
        pc.printf("\n\rSecond number: \n\r");
        pc.scanf("%f", &second_value.value);
    
        send(first_value);
        send(second_value);
        float multiply = first_value.value * second_value.value;
    
        pc.printf("\n\r\n\rLocal: %f * %f = %f\n\r\n\r", first_value.value, second_value.value, multiply);
        float received_result = receive();
        pc.printf("Received result:%f\n\r\n\r", received_result);
        int a;
        pc.scanf("%i", a);
    }
}

```

Slave:
```
#include "mbed.h"
#include <SPISlave.h>
#include <stdio.h>

SPISlave device (PB_5, PB_4, PB_3, PA_4);       // mosi, miso, sclk, ssel
Serial pc (USBTX, USBRX); 

int wait_for_transfer(union float_to_transfer transfer);
int send(union float_to_transfer transfer);

union float_to_transfer {
    float value;
    unsigned char bytes[4];
};

int wait_for_transfer() {
    while(!device.receive()) {}
    switch(device.read()) {
        case 0:
            pc.printf("*** Receive function ***\n\r");
            return 1;
        case 1:
            pc.printf("*** Send function ***\n\r");
            return 2;
        default:
            pc.printf("Don't know what to do\n\r");
    }
    return 0;       
}

/* Receive data from master */
float receive() {
    union float_to_transfer receive_value;   
    int i = 0;
    while(i < 4) {
        while(!device.receive()) {}
        receive_value.bytes[i] = device.read();
        device.reply(receive_value.bytes[i]);
        i++;
    }
    for(int j = 0; j < 4; j++) {
        pc.printf("Byte %i is %02x\n\r", j, receive_value.bytes[j]); 
    }
    return receive_value.value;
}

/* Send data to master */
int send(union float_to_transfer send_value) {
    pc.printf("Sending value: %f\n\r", send_value.value);
    int a[4];
    int i = 0;
    while(i < 4) {
        while(!device.receive()) {pc.printf("a");}
        a[i] = device.read();
        unsigned char byte = send_value.bytes[i];
        pc.printf("Sending: byte %i is %02x\n\r", i, send_value.bytes[i]);
        device.reply(byte);
        i++;
    }
//    for(int j = 0; j < 4; j++) {
//        pc.printf("Byte %i is %02x\n\r", j, a[j]); 
//    }
    return 0;
}

int main() {
    pc.printf("**** SLAVE ****\n\r");
    int i = 0;
    float received_values[2];
    union float_to_transfer result;
    device.reply(0x00);
    while(1) { 
        switch(wait_for_transfer()) {
            case 1:
                received_values[i%2] = receive();
                i++;
                break;
            case 2:
                result.value = received_values[0] * received_values[1];
                send(result); 
                break;
            default:
                pc.printf("Something went wrong");
        }
    }
}
```
Output:

[image]

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-10-01:
@ARMmbed/team-st-mcd Can you review? Is this HAL related issue?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-362*

bcostm, 2018-10-30:
Hi,

Not reproduced with my own test code:
[spi_loop.zip](https://github.com/ARMmbed/mbed-os/files/2530123/spi_loop.zip)

I ran your code but it stopped after "First Number:" ? And I don't see what could be wrong in your code ? Did you progress on your side ? If yes and the problem is solved can you please close this issue ?

Thanks

VVESTM, 2019-06-11:
Hi,
@kubazdz , is this issue still present ?

As mentioned by @bcostm , SPI transfer should be working as expected on STM32 devices.
I see several issues in your source code. The first one is related to SPI management. When slave receives a byte, let's say n, it will send back the n-1 response.  Regarding this, you should call device.reply before device.receive loop in slave send and receive functions.
Then, the SPI slave is full software. It means that you need to ensure that your slave get enough time to prepare the answer between 2 transfers (be careful to printf here for example, or add delay in your master).

Attached is your source code with some modifications. I can test it with a Nucleo L476 as master and Nucleo F429ZI as slave. ([main.zip](https://github.com/ARMmbed/mbed-os/files/3276301/main.zip))

If you agree with this, I propose to close this issue.

jeromecoutant, 2019-06-24:
@0xc0170 Could we close this one ?
Thx

closed 2019-06-24 by 0xc0170