ARMmbed/mbed-os #8267: Overriding 'exit' to Create Immediate and Unconditional System Reset

mattbrown015, 2018-09-27:
### Description
The device we're working on is remote.  If anything goes wrong we really just want it to restart/reset.

We know we can provide a new definition of `mbed_die` because it is declared `WEAK`.

`mbed_die` is called conditionally by '_exit' which is called by `__wrap_exit`.

We would like to make `__wrap_exit`, or `_exit`, just call `system_reset`.

_But_, AFAICT, we can't provide our own definitions of `__wrap_exit` or `_exit` because they are not `WEAK`.

I tried ignoring `mbed_retarget.cpp` but that caused a load of link issues.

Is there anything controversial about making `__wrap_exit` `WEAK`?

Have I missed anything.

I appreciate we could/should also be using a watchdog, I'm just trying to cover all possibilities. 

### Issue request type
[X] Question
[ ] Enhancement
[ ] Bug

pan-, 2018-09-27:
@mattbrown015 `__wrap_exit ` is specific to GCC and not defined for other compilers. Could you explain why overriding `mbed_die` in't enough for your use case ?

mattbrown015, 2018-09-27:
Hi @pan-,

AFAICT, a call to `exit(0)` won't call `mbed_die`.

I don't think `exit(0)` is ever called but I don't want to take the risk.

If I turned a watchdog on it would/could cause while spinning in `exit`.

I might just provide a `mbed_die` definition, it's simple, but I'm also trying to consider all scenarios.

pan-, 2018-09-27:
You're correct, `exit(0)` won't call `mbed_die`. 

Looking at the current code, `exit` behaviour is not really well defined as an example, running threads are not cleaned up and other things are happening which are not fully standard to save resources.

That would be useful to have termination as defined as boot is and includes valuable extension points. 

@bulislaw @kjbracey-arm @c1728p9 What do you think ?

kjbracey-arm, 2018-09-27:
To be 100% solid here, you would want a watchdog; and you'd make sure exit did actually trigger it.

You should also be able to force an immediate reset with `NVIC_SystemReset`.

There are a bunch of issues about what to do on shutdown, and we tend to actually want to try to skip standard behaviour, to reduce image size - eg I'd be inclined to skip calling destructors to save static objects pulling in code, and we have options to skip stream flushing. With that point of view, I think we currently tend to think of any exiting of main as not normal behaviour, even if the return code is 0.

Given that, it would make sense to have the default exit or abort behaviour to lead to an mbed die or error  that announces on the console that it happened.

mattbrown015, 2018-09-27:
We've been thinking about this today and have decided to provide our own definition of `mbed_die` that simply calls `system_reset`.

We think this means `MBED_ASSERT`, `exit(1)` and `abort()` will all result in a reset.

`exit(0)` will result in a spin. This isn't ideal for our application but feels like the correct default approach for a library like Mbed OS.

Whether or not Mbed OS should provide further means of customisation is a trickier question.  Making `__wrap_exit` and/or `_exit` `WEAK` would work for us but we can see that it is complicated by the need to support multiple toolchains.

I appreciate what we really need is a watchdog but turning on the watchdog has plenty of gotchas.

SenRamakri, 2018-09-27:
Hi @mattbrown015,
When something goes wrong, the system should either crash(like hardfault) or call the error routine.
Currently all such fatal system errors are routed though mbed_error() call (or error() call in some legacy cases). Both of them will end up calling an application hook if one is registered by the application. So, my suggestion is to use an application specific hook to catch the error and do a reset as required. As pointed above, things like __wrap_exit are toolchain specific and doing the same for other toolchains create future maintenance issues in your codebase. Please see the below document for error handling/error hook. https://os.mbed.com/docs/v5.10/apis/error-handling.html

mattbrown015, 2018-09-28:
Hi @SenRamaki,

We're getting off the point now but it's interesting (well I think so)...

We haven't embraced [error handling](https://os.mbed.com/docs/v5.10/apis/error-handling.html), in part at least, because it wasn't so visible when we started our project back in V5.9.  We have used `MBED_ASSERT` a great deal.  I think it is too late for us to change our coding practices now.

The LoRaWAN stack also uses `MBED_ASSERT` and I fired one of its asserts only the other day.  Hence why I'm interested in what will happen in the field if an assert fires.  We've been talking about defining `NDEBUG` but having fired an assert recently I think we should keep the asserts.

Obviously the Mbed OS code we're using may well call `error()` and, AFAICT, this will call `exit(1)` and in turn our `mbed_die` definition.  In other words, I happy that calling `error()` will cause a restart for us.

Clearly my comments haven't 'struck a chord' with anyone so I'll close the question. :-)

Thanks,
Matt

mattbrown015, 2018-09-28:
I meant to click 'Close' with my previous comment!

closed 2018-09-28 by mattbrown015