ARMmbed/mbed-os #8278: Low power ticker wrapper layer still used when suspended

fkjagodzinski, 2018-09-28:
### Description

The `LowPowerTickerWrapper` class is still used to handle lp ticker overflows even after it has been disabled with a call to `lp_ticker_wrapper_suspend()`.

https://github.com/ARMmbed/mbed-os/blob/f1587b50d7d1e36e3536a88dd92884029a1f4f10/hal/mbed_lp_ticker_wrapper.h#L52-L57

As a result, the us ticker which is used internally by the wrapper, periodically locks deepsleep sleep mode.

#### Targets
`NUCLEO_F746ZG`

#### Mbed OS SHA
b1b6399 -- current master

#### Steps to reproduce
Check out 76fc2922ae29e0f0feea2c131fc52fd51f250a77 (https://github.com/fkjagodzinski/mbed-os/commit/76fc2922ae29e0f0feea2c131fc52fd51f250a77) and use a logic analyser to capture data from `D0`, `D1`, `D2` & `D3`.

[image]

[image]

[lp_ticker_wrapper-logic_dump.zip](https://github.com/ARMmbed/mbed-os/files/2427987/lp_ticker_wrapper-logic_dump.zip)

---

The high level result of this issue can be seen here:
```cpp
#include "mbed.h"
#include "mbed_lp_ticker_wrapper.h"

#define SLEEP_DURATION_US 100000ULL
#define SERIAL_FLUSH_TIME_MS 20

void wakeup_callback(volatile int *wakeup_flag)
{
    (*wakeup_flag)++;
}

int main()
{
    // Suspend the RTOS kernel scheduler to prevent interference with duration of sleep.
    osKernelSuspend();
    // Suspend the low power ticker wrapper to prevent interference with deep sleep lock.
    lp_ticker_wrapper_suspend();

    volatile int wakeup_flag;
    LowPowerTimeout lp_timeout;

    bool deepsleep_status1 = sleep_manager_can_deep_sleep();
    sleep_manager_lock_deep_sleep();
    bool deepsleep_status2 = sleep_manager_can_deep_sleep();

    wakeup_flag = 0;
    lp_timeout.attach_us(mbed::callback(wakeup_callback, &wakeup_flag), SLEEP_DURATION_US);

    while (wakeup_flag == 0) {
        sleep_manager_sleep_auto();
    }

    bool deepsleep_status3 = sleep_manager_can_deep_sleep();
    sleep_manager_unlock_deep_sleep();
    bool deepsleep_status4 = sleep_manager_can_deep_sleep();

    printf("------------------------------------------------------------\r\n");
    printf("deepsleep_status1 = %i, %i expected\r\n", deepsleep_status1, 1);
    printf("deepsleep_status2 = %i, %i expected\r\n", deepsleep_status2, 0);
    printf("deepsleep_status3 = %i, %i expected\r\n", deepsleep_status3, 0);
    printf("deepsleep_status4 = %i, %i expected\r\n", deepsleep_status4, 1);
    printf("\r\n");
    wait_ms(SERIAL_FLUSH_TIME_MS);
}
```
```bash
mbed compile -t GCC_ARM -m NUCLEO_F746ZG -f --sterm
```
Output:
```bash
------------------------------------------------------------
deepsleep_status1 = 1, 1 expected
deepsleep_status2 = 0, 0 expected
deepsleep_status3 = 0, 0 expected
deepsleep_status4 = 0, 1 expected
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug  

CC @c1728p9 @mprse @jamesbeyond

c1728p9, 2018-10-02:
This may not be a bug with the low power ticker wrapper. To prevent interrupts you'll need to disable all the users of the LowPowerTicker wrapper before you suspend the low power ticker wrapper, otherwise the behavior is undefined. In this case its the common ticker layer which is still active and causing the interrupts you are seeing. If you suspend the common layer with a call to `ticker_suspend(get_lp_ticker_data());` then you should no longer see locked deep sleep.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-363*

c1728p9, 2018-10-16:
> Our docs are missing the part about undefined behaviour. Shall I add a note with a warning here?

Yeah, please add this info.

> However, I expected these interrupts to pass through unconditionally, without using the us ticker and locking a deepsleep mode.

The current implementation pass these interrupts through unconditionally. Its just calling into the ticker warpper layer when it is suspended which causes undefined behavior.

PR #8279 (2018-10-26): Fix LowPowerTickerWrapper operation when suspended
### Description
Update the `LowPowerTickerWrapper` class logic to stop using the internal
`Timeout` object for scheduling LP ticker interrupts after the wrapper has
been suspended.

Fixes #8278

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change

CC @c1728p9 @mprse @jamesbeyond
commit 023603fa (2018-10-26): Fix LowPowerTickerWrapper operation when suspended

Update the LowPowerTickerWrapper class logic to stop using its internal
Timeout object for scheduling LP ticker interrupts after the wrapper has
been suspended.

Fixes #8278
closed 2018-10-26 by cmonr