ARMmbed/mbed-os #8282: Mbed 2 BLE devices could not build by online IDE

toyowata, 2018-09-29:
### Description
Reported here: https://os.mbed.com/forum/ja/topic/33299/

I got build error below when I used Mbed 2 BLE devices.

```
Error: The build system did not finish successfully.
Info: Unable to download. Fix the reported errors...
```

All board below failed to build by mbed_blinky example.

* HRM1017
* nRF51822
* Wallbot BLE
* BBC micro:bit

It seems all boards below failed to build:
https://os.mbed.com/platforms/?mbed-os=17&connectivity=3

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-10-01:
cc @ARMmbed/mbed-os-pan @ARMmbed/mbed-os-tools

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-515*