ARMmbed/mbed-os #8289: Creating a non-initialised linker region with ARM toolchain

RobMeades, 2018-10-01:
### Description
I need to create a region of RAM that is not initialised at power-on (to contain logging data that must survive across a reset).  I can do this with `GCC_ARM` and `IAR` but have completed failed to make it work with `ARM` toolchain.  Could the merge process to bring together APP and non-APP images be somehow affecting the way my image is being initialised?

I have made my own scatter-link file region, marked as `UNINIT`, and, in my `__attribute__` call, I am specifying that this is `zero_init` RAM (otherwise I'm told that the ARM linker thinks it is `Data` RAM and zero-initialises it no matter what you say). For my test code below the `.map` file shows that the linker _is_ putting my test variable into the correct region:

```
Execution Region RW_IRAM1 (Base: 0x200032ac, Size: 0x00000004, Max: 0x00000400, ABSOLUTE, UNINIT)
Base Addr    Size         Type   Attr      Idx    E Section Name  Object
0x200032ac   0x00000004   Zero   RW         8    .bss.noinit      main.o
```

Yet with this test code:

```
#include "mbed.h"

// An unsigned int in an uninitialised RAM area
__attribute__ ((section(".bss.noinit"), zero_init))
unsigned int gRetained;

// Entry point
int main()
{
    printf("Retained RAM variable is %d.\n", gRetained);
    gRetained++;
    printf("Retained RAM variable incremented to %d.\n", gRetained);
    printf("Resetting...\n");
    wait_ms(1000);
    NVIC_SystemReset();
}
```
...the output I get is:

```
Retained RAM variable is 0.
Retained RAM variable incremented to 1.
Resetting...
Retained RAM variable is 0.
Retained RAM variable incremented to 1.
Resetting...
Retained RAM variable is 0.
...
```

Here is my scatter-link file in full where `RW_IRAM1` is the region I'm talking about:

```
#! armcc -E

/* Default to no softdevice */
#if !defined(MBED_APP_START)
  #define MBED_APP_START 0x0
#endif

#if !defined(MBED_APP_SIZE)
  #define MBED_APP_SIZE 0x80000
#endif

/* Physical RAM */
#define MBED_RAM_PHYSICAL_START 0x20000000
#define MBED_RAM_PHYSICAL_SIZE 0x10000

/* Reserved areas */
#define MBED_RAM_SOFT_DEVICE_SIZE 0x31d0
#define MBED_RAM_UNINIT_AREA_SIZE 1024

/* If app_start is 0, do not set aside space for the softdevice */
#if MBED_APP_START == 0
  #define MBED_RAM_START  MBED_RAM_PHYSICAL_START
  #define MBED_RAM_SIZE   MBED_RAM_PHYSICAL_SIZE
#else
  #define MBED_RAM_START  (MBED_RAM_PHYSICAL_START + MBED_RAM_SOFT_DEVICE_SIZE)
  #define MBED_RAM_SIZE   (MBED_RAM_PHYSICAL_SIZE - MBED_RAM_SOFT_DEVICE_SIZE)
#endif

#define MBED_RAM0_START MBED_RAM_START
#define MBED_RAM0_SIZE  0xDC
#define MBED_RAM1_START (MBED_RAM0_START + MBED_RAM0_SIZE)
#define MBED_RAM1_SIZE  MBED_RAM_UNINIT_AREA_SIZE
#define MBED_RAM2_START (MBED_RAM1_START + MBED_RAM1_SIZE)
#define MBED_RAM2_SIZE  (MBED_RAM_SIZE - MBED_RAM0_SIZE - MBED_RAM1_SIZE)

LR_IROM1 MBED_APP_START MBED_APP_SIZE {
  ER_IROM1 MBED_APP_START MBED_APP_SIZE {
    *.o (RESET, +First)
    *(InRoot$$Sections) 
    .ANY (+RO)
  }

  RW_IRAM0 MBED_RAM0_START UNINIT MBED_RAM0_SIZE { ;no init section
    *(*nvictable)
  }
  RW_IRAM1 MBED_RAM1_START UNINIT MBED_RAM1_SIZE { ;no init section
    *(*noinit)
  }
  RW_IRAM2 MBED_RAM2_START MBED_RAM2_SIZE {
    .ANY (+RW +ZI)
  }
}
```

This is on an `NRF52832` target (`UBLOX_EVK_NINA_B1`). `ARMCC` version is 5.06 update 6 (build 750).  mbed-os version is `40e9f41b4` "Merge pull request #7821 from klaas019/master".

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

RobMeades, 2018-10-03:
My understanding of the situation has improved now and I believe the problem is down to how BLE configuration works on this device.  ARM tools support have confirmed that my linker configuration is correct and that nothing is statically overwriting my `noinit` zone.  However, Nordic SoftDevice is enabled in my build and, though BLE is not currently being run, I suspect that something somewhere at boot is writing BLE configuration data at the `APP_RAM_START` location, which is where my region is.  If I extend my reserved region to 8 kbytes, set it all to `0xFF` and print it out again after a reset, you can clearly see that everything up to 0x20003b54 is being overwritten:

```
0x20003b2c: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x20003b3c: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x10 0x00 
0x20003b4c: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 
0x20003b5c: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 
0x20003b6c: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 
```

So I can't create a RAM region of my own before the usual RAM area as the BLE functionality in mbed assumes it has control of this.  What I'd really like to do is put my `noinit` region _after_  the usual RAM area and before the heap, but, as far as I know, the ARM linker provides no mechanism to do that.

Any clues as to how I get this done?

screamerbg, 2018-10-03:
CC @pan- @marcuschangarm

RobMeades, 2018-10-04:
And of course, the other question is, with the scatter-link file as it stands, are the first kbyte or so of variables in RW memory being overwritten by BLE config, or (more likely) vice-versa...?

RobMeades, 2018-10-04:
Just FYI, if I create a target which inherits `UBLOX_EVK_NINA_B1` but removes Soft Device (by removing the labels `SOFTDEVICE_S132_FULL` and `SOFTDEVICE_COMMON` while adding the label `SOFTDEVICE_NONE`) then this RAM overwrite does not occur.  So it is BLE of config data or something to do with Soft Device that is making assumptions about the availability of RAM immediately beyond the vector table.

AGlass0fMilk, 2019-01-14:
The master boot record and the soft device assume control of a chunk of memory at the beginning of the RAM address space. Check out the soft device [specification](url
https://www.nordicsemi.com/-/media/DocLib/Other/SoftDevice_Spec/S132SDSv62pdf.pdf?la=en#page67) for more info on resource requirements if you haven’t already.

You could add an uninitialized RAM region where the application RAM currently starts, and move the application RAM region up however many kB your retention region requires.

I’m not sure how to achieve this with the ARM tool chain, but with GCC ARM it would be a quick linker script change.

Hope that helps.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@RobMeades Please reopen this issue if you still have questions.