ARMmbed/mbed-os #8290: Nucleo_blink_led example export to Embitz for Nucleo-F411RE fails to build

aidanoconnor, 2018-10-01:
### Description
Created new program in mbed online compiler using Nucleo_blink_led template for target NUCLEO-F411RE and exported to EmBitz v1.11. Build failed in EmBitz as preprocessor syntax in linker script not recognized.

```
-------------- Build: Nucleo_blink_led in Nucleo_blink_led ---------------

Linking executable: .\build\Nucleo_blink_led.elf
c:/program files (x86)/embitz/1.11/share/em_armgcc/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe:mbed/TARGET_NUCLEO_F411RE/TOOLCHAIN_GCC_ARM/STM32F411XE.ld:1: ignoring invalid character `#' in expression
c:/program files (x86)/embitz/1.11/share/em_armgcc/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe:mbed/TARGET_NUCLEO_F411RE/TOOLCHAIN_GCC_ARM/STM32F411XE.ld:1: syntax error
collect2.exe: error: ld returned 1 exit status
Process terminated with status 1 (0 minutes, 1 seconds)
2 errors, 0 warnings (0 minutes, 1 seconds)
```
```
#if
 !defined(MBED_APP_START)
  #define MBED_APP_START 0x08000000
#endif

#if !defined(MBED_APP_SIZE)
  #define MBED_APP_SIZE 512K
#endif

/* Linker script to configure memory regions. */
MEMORY
{
  FLASH (rx)   : ORIGIN = MBED_APP_START, LENGTH = MBED_APP_SIZE
  RAM (rwx)  : ORIGIN = 0x20000198, LENGTH = 128k - 0x198
}
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

jeromecoutant, 2018-10-01:
Hi
I don't know EmBitz, but I got a question about this part :

> gcc/arm-none-eabi/5.4.1/

See https://os.mbed.com/docs/v5.10/tools/index.html
GCC version should be 6.x

aidanoconnor, 2018-10-01:
@jeromecoutant Thanks for pointing out the version requirement.

I have re-checked with GCC v6.3.1 with the same results.
I think if the linker script uses the preprocessor it would have to be invoked manually as a separate build step.

```
-------------- Build: Nucleo_blink_led in Nucleo_blink_led ---------------

Compiling: main.cpp
Linking executable: .\build\Nucleo_blink_led.elf
c:/analog devices/crosscore embedded studio 2.8.0/arm/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe:mbed/TARGET_NUCLEO_F411RE/TOOLCHAIN_GCC_ARM/STM32F411XE.ld:1: ignoring invalid character `#' in expression
c:/analog devices/crosscore embedded studio 2.8.0/arm/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe:mbed/TARGET_NUCLEO_F411RE/TOOLCHAIN_GCC_ARM/STM32F411XE.ld:1: syntax error
collect2.exe: error: ld returned 1 exit status
Process terminated with status 1 (0 minutes, 4 seconds)
2 errors, 0 warnings (0 minutes, 4 seconds)
```

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-10*

jeromecoutant, 2018-10-05:
Some "preprocessing" should miss on this ld file...
I can't help

theotherjimmy, 2018-10-23:
@aidanoconnor Do you happen to know how to add a separate build step? I can add one that preprocesses the linker script then.

aidanoconnor, 2018-10-23:
@theotherjimmy In the EmBitz project _Build Options_ dialog, in the _Pre/post build steps_ tab, you could add the appropriate command or script in the _Pre-build steps_ section.

elm3, 2018-11-12:
Hello Aidan,

Have you had any luck with this issue?  did the suggestion by @theotherjimmy help?

Thanks,
Earl

jeromecoutant, 2019-01-29:
Should we close this issue ?
Thx