ARMmbed/mbed-os #8293: What is the requirement for ESP8266 firmware version?

coisme, 2018-10-02:
### Description

I'm trying to use ESP8266 with Mbed OS 5.10. The [README.md](https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/README.md)  says:

> ESP8266 modules come in different shapes and formats, but the most important factor is the firmware version in it. To make sure that the firmware in your module is compatible with Mbed OS, follow [the Update guide](https://os.mbed.com/teams/ESP8266/wiki/Firmware-Update).

However, [the Update guide](https://os.mbed.com/teams/ESP8266/wiki/Firmware-Update) doesn't include information about supported firmware version. What firmware version is supported by Mbed OS 5.10?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

yennster, 2018-10-10:
@sarahmarshy Do you happen to know which specific firmware version numbers are supported in Mbed OS? Per your esp8266 update guide: https://os.mbed.com/teams/ESP8266/wiki/Firmware-Update

sarahmarshy, 2018-10-10:
v2.0 I believe. @kegilbert?

kegilbert, 2018-10-10:
Good question, I haven't touched the ESP8266 driver myself for a bit. @VeijoPesonen seems to have been maintaining the repo the most, although we have pulled the driver itself into the components directory of Mbed OS. I'll ask around for a more clear owner.

VeijoPesonen, 2018-10-11:
@coisme as @kegilbert mentioned I have been updating the driver lately. The group looking after the driver is @ARMmbed/mbed-os-ipcore - which I'm part of. The in-tree(Mbed OS) driver is basically the same as [v1.6](https://github.com/ARMmbed/esp8266-driver/tree/v1.6) found from [esp8266-driver](https://github.com/ARMmbed/esp8266-driver) repository. What has come after v1.6 can only be found from the master branch of the external repository.

v1.6 driver supports AT firmware up to version 1.6.2 - please see [ESP8266 Resources](https://www.espressif.com/en/products/hardware/esp8266ex/resources). I.e., Mbed OS 5.10 supports AT firmwares up to version 1.6.2. Master branch allows usage of AT firmware version 1.7.0.0.
*NOTE* I'm talking here about AT FW version - not SDK version. SDK FW without AT is no use to us.
*NOTE* AT firmware version numbers and driver version numbers do not match to each other. E.g., driver version v1.7 does not mean directly that it will support AT FW version 1.7

The external repository is to be deprecated at some point but for now I continue development on that. Once deprecated, changes found only from the external repository will be merged to the in-tree driver.

coisme, 2018-10-11:
@VeijoPesonen Got it! Thank you. So I should see the `AT` tab, not `SDK & Demos`, and check the version there.

closed 2018-10-11 by coisme