ARMmbed/mbed-os #8294: LPC546XX: hard faults in various tests

0xc0170, 2018-10-02:
### Description

LPC546XX-GCC_ARM.tests-netsocket-tcp.TCPSOCKET_ECHOTEST_NONBLOCK (from LPC546XX-GCC_ARM) test :

```
[1538422528.68][CONN][RXD] >>> Running case #2: 'TCPSOCKET_ECHOTEST_NONBLOCK'...
[1538422528.76][CONN][INF] found KV pair in stream: {{__testcase_start;TCPSOCKET_ECHOTEST_NONBLOCK}}, queued...
[1538422528.84][CONN][RXD] [Sender#00] bytes sent: 1
[1538422529.82][CONN][RXD] [Recevr#00] bytes received: 1
[1538422529.82][CONN][RXD] [Sender#01] bytes sent: 2
[1538422530.82][CONN][RXD] [Recevr#01] bytes received: 2
[1538422530.92][CONN][RXD] [Sender#02] bytes sent: 3
[1538422531.92][CONN][RXD] [Recevr#02] bytes received: 3
[1538422531.92][CONN][RXD] [Sender#03] bytes sent: 4
[1538422532.93][CONN][RXD] [Recevr#03] bytes received: 4
[1538422532.93][CONN][RXD] [Sender#04] bytes sent: 5
[1538422533.93][CONN][RXD] [Recevr#04] bytes received: 5
[1538422533.93][CONN][RXD] [Sender#05] bytes sent: 6
[1538422534.94][CONN][RXD] [Recevr#05] bytes received: 6
[1538422534.94][CONN][RXD] [Sender#06] bytes sent: 7
[1538422535.94][CONN][RXD] [Recevr#06] bytes received: 7
[1538422536.04][CONN][RXD] [Sender#07] bytes sent: 8
[1538422537.04][CONN][RXD] [Recevr#07] bytes received: 8
[1538422537.04][CONN][RXD] [Sender#08] bytes sent: 9
[1538422538.04][CONN][RXD] [Recevr#08] bytes received: 9
[1538422538.04][CONN][RXD] [Sender#09] bytes sent: 10
[1538422538.55][CONN][RXD] 
[1538422538.55][CONN][RXD] ++ MbedOS Fault Handler ++
[1538422538.55][CONN][RXD] 
[1538422538.55][CONN][RXD] FaultType: HardFault
[1538422538.55][CONN][RXD] 
[1538422538.55][CONN][RXD] Context:
[1538422538.64][CONN][RXD] R0   : 00000000
[1538422538.64][CONN][RXD] R1   : 20002DF4
[1538422538.64][CONN][RXD] R2   : 20003220
[1538422538.65][CONN][RXD] R3   : 000000E8
[1538422538.65][CONN][RXD] R4   : 2000D850
[1538422538.65][CONN][RXD] R5   : 20005568
[1538422538.74][CONN][RXD] R6   : 20010844
[1538422538.74][CONN][RXD] R7   : 00000008
[1538422538.74][CONN][RXD] R8   : 00000000
[1538422538.75][CONN][RXD] R9   : 00000000
[1538422538.75][CONN][RXD] R10  : 00000000
[1538422538.75][CONN][RXD] R11  : 00000000
[1538422538.75][CONN][RXD] R12  : 00014185
[1538422538.85][CONN][RXD] SP   : 200107B4
[1538422538.85][CONN][RXD] LR   : 000143B3
[1538422538.85][CONN][RXD] PC   : 00000000
[1538422538.85][CONN][RXD] xPSR : 40000200
[1538422538.85][CONN][RXD] PSP  : 20010790
[1538422538.95][CONN][RXD] MSP  : 20027FC0
[1538422538.95][CONN][RXD] CPUID: 410FC241
[1538422538.95][CONN][RXD] HFSR : 40000000
[1538422538.95][CONN][RXD] MMFSR: 00000000
[1538422538.96][CONN][RXD] BFSR : 00000000
[1538422538.96][CONN][RXD] UFSR : 00000002
[1538422538.96][CONN][RXD] DFSR : 00000008
[1538422539.05][CONN][RXD] AFSR : 00000000
[1538422539.05][CONN][RXD] Mode : Thread
[1538422539.05][CONN][RXD] Priv : Privileged
[1538422539.05][CONN][RXD] Stack: PSP
[1538422539.05][CONN][RXD] 
[1538422539.05][CONN][RXD] -- MbedOS Fault Handler --
[1538422539.05][CONN][RXD] 
[1538422539.05][CONN][RXD] 
[1538422539.05][CONN][RXD] 
[1538422539.15][CONN][RXD] ++ MbedOS Error Info ++
[1538422539.15][CONN][RXD] Error Status: 0x80FF013D Code: 317 Module: 255
[1538422539.15][CONN][RXD] Error Message: Fault exception
[1538422539.25][CONN][RXD] Location: 0x12D47
[... 156 lines cut ...]
[1538418633.30][CONN][RXD] 
[1538418633.30][CONN][RXD] 
[1538418633.30][CONN][RXD] ++ MbedOS Error Info ++
[1538418633.40][CONN][RXD] Error Status: 0x80FF013D Code: 317 Module: 255
[1538418633.40][CONN][RXD] Error Message: Fault exception
[1538418633.40][CONN][RXD] Location: 0x5B27
[1538418633.40][CONN][RXD] Error Value: 0x4DD6
[1538418633.50][CONN][RXD] Current Thread: Id: 0x200026A4 Entry: 0x5C73 StackSize: 0x1000 StackMem: 0x200026E8 SP: 0x20027F28 
[1538418633.60][CONN][RXD] For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF013D
[1538418633.70][CONN][RXD] -- MbedOS Error Info --
[1538418853.93][HTST][INF] test suite run finished after 240.37 sec...
[1538418853.93][CONN][INF] received special event '__host_test_finished' value='True', finishing
```

This is currently ongoing on master, should be able to reproduce this fetching the master.

For internal reference: IOTTESTINF-4055

@ARMmbed/team-nxp @ARMmbed/mbed-os-ipcore @studavekar 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

mmahadevan108, 2018-10-02:
@kjbracey-arm There an updated DAPLink firmware available that fixes a bug where the entire binary was not getting programmed to flash.
https://github.com/ARMmbed/DAPLink/commit/3ae822fa850f2f075d9cffb4982de49f36527c76

The Updated interface firmware for LPC546XX is available at the below link:
https://os.mbed.com/teams/NXP/wiki/Updating-LPCXpresso-firmware

Can you please update and check if this is still an issue.

0xc0170, 2018-10-02:
@mmahadevan108 The fix you referenced was only for images >256k or also smaller images might be affected ? Because our test images we do in morph test are not that big.

kjbracey-arm, 2018-10-02:
The ticker crash is a bit inconclusive. The microsecond timer event head (in `events` from mbed_us_ticker_api.c) is corrupt.

Can't see any flaw in the `Ticker` test or `Ticker` itself. The tests are a big aggressive at creating `Ticker`s on the stack and expecting the destructor to clean up, but looks like it should be working.  Only quirk I see is that `TimerEvent::remove` gets called twice via the inherited destructors, but the second one should be a no-op.

If I can find the board here I can try to reproduce locally under a debugger tomorrow.

mmahadevan108, 2018-10-02:
I ran all the mbed tests. I am not able to reproduce this crash on the LPC54628 and LPC54608 boards.

kjbracey-arm, 2018-10-03:
@c1728p9 has ascertained that one or more boards in the CI has worn-out flash, which causes unreliable reads. So I think we suspend investigation of the crashes. The serial loss issue remains.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-9*

closed 2018-10-23 by ciarmcom