ARMmbed/mbed-os #8296: CI: update STLINK FW firmware upgrade version tool

jeromecoutant, 2018-10-02:
### Description

CI log extract:

````
TID = 08160221052A601C3A6EF937
WARNING:mbedls.lstools_base:Could not read htm on from usb id 0674FF525455777267013160. Falling back to usb id
WARNING:mbedls.lstools_base:Could not read htm on from usb id 066DFF525455777267011718. Falling back to usb id
updating STLINK for 08160221052A601C3A6EF937:0676FF535452775187163159...
java -jar /usr/local/bin/STLinkUpgrade/STLinkUpgrade.jar -sn 0676FF535452775187163159 -force_prog
Firmware version detected: V2J30M19
....................Upgrade is successful.
Version read: 2.30.19
interface firmware version : 0221 
````

Please update FW version to  **2.32.22**

https://www.st.com/en/development-tools/stsw-link007.html

@schstm @bcostm 

Thx

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-10-02:
Thanks @jeromecoutant for the report. I'll pass this to the test team

@ARMmbed/mbed-os-test-team

OPpuolitaival, 2018-10-02:
@jeromecoutant Why we need to update this?

OPpuolitaival, 2018-10-02:
Internal ticked id: IOTTESTINF-4071

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-8*

studavekar, 2018-10-04:
Thanks, @jeromecoutant will update it to 2.32.22

studavekar, 2018-10-12:
> Thanks, @jeromecoutant will update it to 2.32.22 for CI in Austin.

updated is complete in austin CI.

closed 2019-10-13 by ciarmcom