ARMmbed/mbed-os #8297: CI Exporters: failing uvision5

0xc0170, 2018-10-02:
### Description

One of the jobs that recently failed, different targets, same toolchain (uvision5):
http://mbed-os.s3-website-eu-west-1.amazonaws.com/?prefix=builds/exporter/8072/ . From this report is obvious it started building  but failed somewhere. From CI logs, there is a traceback, more investigation needed. Please review the exporters log.

@ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-test 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-10-02:
This is from CI

```
10:33:45 Traceback (most recent call last):
10:33:45   File "../mbed-os/tools/test/examples/examples.py", line 133, in <module>
10:33:45     sys.exit(main())
10:33:45   File "../mbed-os/tools/test/examples/examples.py", line 84, in main
10:33:45     return args.fn(args, config, examples)
10:33:45   File "../mbed-os/tools/test/examples/examples.py", line 90, in do_export
10:33:45     results = lib.export_repos(config, args.ide, args.mcu, examples)
10:33:45   File "C:\builds\ws\exporter-build-matrix\2802\DISCO_L475VG_IOT01A_uvision5\sources\mbed-os\tools\test\examples\examples_lib.py", line 297, in export_repos
10:33:45     if EXPORTERS[ide].build(example_project_name, cleanup=False):
10:33:45   File "C:\builds\ws\exporter-build-matrix\2802\DISCO_L475VG_IOT01A_uvision5\sources\mbed-os\tools\export\uvision\__init__.py", line 294, in build
10:33:45     with open(log_name, 'r') as f:
10:33:45 IOError: [Errno 2] No such file or directory: 'build_log.txt'
```

@OPpuolitaival @studavekar @theotherjimmy  Any suggestions? 

I can't reproduce this locally (using the same example_lib command I get 9 errors). I suspect this might be due to cmsis pack, so checking uvision5 export:

`mbed export -m DISCO_L475VG_IOT01A -i uvision5` , using uvision 5.24 - it requests device pack that I have installed. Once I install, still complains that L475VG is not available anyway and I am in the circle - I selected one of  L475VGx targets, the local blinky compiles then. Did cmsis pack change ? I tried uninstall L4 cmsis pack , did not help.

@ARMmbed/team-st-mcd Are you able to export for this target for uvision5?

0xc0170, 2018-10-02:
Looking at this closer - STM32L4xx DFP 2.0.0 is requested for me locally and does not fit the target definiiton in case for L475. Looking at other failures, all failures we have now in CI are related to this DFP. Did something corrupt since the update - cmsis pack was updated yesterday

@miklis

0xc0170, 2018-10-02:
@ARMmbed/mbed-os-test Need to review this one ticket IOTTESTINF-3857

bcostm, 2018-10-02:
> @ARMmbed/team-st-mcd Are you able to export for this target for uvision5?

Yes it works for me. I have the same L4 pack version as you.

0xc0170, 2018-10-02:
[image]

[image]

[image]

Are you able to build the blinky project? From the targets taht fail in our exporters CI, they are showing the same errors when I export and open in uvision 5.24. Restarting entire machine just in case

0xc0170, 2018-10-02:
Attaching uvision build log, using `python -u ./mbed-os/tools/test/examples/examples.py export uvision5 --mcu DISCO_L475VG_IOT01A` command (having mbed-os latest master, plus blinky in the same folder , from this folder run this script):

[build_log.txt](https://github.com/ARMmbed/mbed-os/files/2438311/build_log.txt)

bcostm, 2018-10-02:
I have the same "Device not found" error when the project is loaded. I have selected the STM32L475VG**Tx** in the Options>Device tab and the build was OK.

But I don't have the "__UVISION_VERSION SETA 524" error ? I use uvision V5.24.2.0.

0xc0170, 2018-10-02:
> I have the same "Device not found" error when the project is loaded. I have selected the STM32L475VGTx in the Options>Device tab and the build was OK.

If I fix the MCU, build is OK. If I don't I get that error .this is however locally , still does not fully explain what is going on with L4 targets in export and why they started failing. PR 8301 shows that process fails and returns windows error.

0xc0170, 2018-10-04:
> I have the same "Device not found" error when the project is loaded. I have selected the STM32L475VGTx in the Options>Device tab and the build was OK.

@bcostm is this common behavior? Shouldn't the device be set by the exporters (the correct version). I am just trying to minimize the area for an error. The latest result in 8301 uncover windows returning error when building STM32 L4 targets, something goes wrong with uvision using the examples_lib.py script

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-7*

0xc0170, 2018-10-04:
The uvision reports this

```
[mbed] Auto-installing missing Python modules...
SUCCESS exporting mbed-os-example-blinky NUCLEO_L486RG uvision5
Building mbed-os-example-blinky NUCLEO_L486RG uvision5
Log file missing! 
Out:

Err:
Error code is -1073741819! 
```

The error code indicates invalid access as I understand.

bcostm, 2018-10-05:
> Is this common behavior? 

No

> Shouldn't the device be set by the exporters (the correct version).

Yes. The device for the Keil exporter is taken (I think) from the `device_name` field in targets.json. As this field is equal to STM32L475VG and not STM32L475VGTx we have this problem...

Note that for IAR exporter the device names are defined in a specific `iar_definition.json` file.

This is also the same for sw4stm32 exporter where the device names are defined in the `sw4stm32/__init__.py` file.

So, I think we have either to add a uvision_definition.json file or to change the device_name in targets.json. But in this second case this will break the IAR and sw4stm32 exporters as the device name will have to be changed also... I don't know if I'm clear...

0xc0170, 2018-10-05:
> So, I think we have either to add a uvision_definitions.json file or to change the device_name in targets.json. But in this second case this will break the IAR exporter as the device name will have to be changed also... I don't know if I'm clear...

I also was told that latest uvision does this MCU matching so should not be a problem. I tested this and can compilied from the command line using 5.26 (previously I had 5.24 on one computer). Therefore 3rd option would be to update the CI uvision. I'll talk to the CI team

0xc0170, 2018-10-05:
We updated to uvision 5.26 and exporters are green again (last 3 runs had uvision5 green!)

This should be now fixed. @bcostm I recommend updating as well to your local machines due to the feature described above

closed 2018-10-05 by 0xc0170