ARMmbed/mbed-os #8300: Configuration options for STM_EMAC buffer counts

juhoeskeli, 2018-10-02:
### Description

On boards (e.g. NUCLEO_F207ZG) where there is only 128kB of RAM it can be quite a tight fit in some cases. One place where it is possible to achieve memory savings is EMAC driver. There the TX and RX buffer counts can be adjusted with little to no ill effects but still significant memory savings (relative). 

I did the following manually:

`diff --git a/targets/TARGET_STM/TARGET_STM32F2/device/stm32f2xx_hal_conf.h b/targets/TARGET_STM/TARGET_STM32F2/device/stm32f2xx_hal_conf.h
index 23c209c..027fc1f 100644
— a/targets/TARGET_STM/TARGET_STM32F2/device/stm32f2xx_hal_conf.h
+++ b/targets/TARGET_STM/TARGET_STM32F2/device/stm32f2xx_hal_conf.h
@@ -175,7 +175,7 @@
/* Definition of the Ethernet driver buffers size and count */ 
#define ETH_RX_BUF_SIZE ETH_MAX_PACKET_SIZE /* buffer size for receive */
#define ETH_TX_BUF_SIZE ETH_MAX_PACKET_SIZE /* buffer size for transmit */
-#define ETH_RXBUFNB 4U /* 4 Rx buffers of size ETH_RX_BUF_SIZE */
+#define ETH_RXBUFNB 1U /* 4 Rx buffers of size ETH_RX_BUF_SIZE */
#define ETH_TXBUFNB 4U /* 4 Tx buffers of size ETH_TX_BUF_SIZE */

/* Section 2: PHY configuration section */
`
This saves additional 4672 bytes of RAM while preserving the functionality in our use case.

For example on how to make this configurable see https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_Freescale_EMAC/mbed_lib.json. I would propose same kind of mechanism for STM_EMAC driver. In case the configuration option is implemented the user could adjust the buffer counts in their mbed_app.json.

### Issue request type

[ ] Question
[X] Enhancement 
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-6*

yogpan01, 2018-10-11:
@screamerbg Can you please take this up with ST as this is preventing us to add support for NUCLEO-F207ZG.

jeromecoutant, 2018-10-11:
@yogpan01 feel free to propose a PR
Regards,
Jerome

yogpan01, 2018-10-11:
@JaniSuonpera When you are verifying the patch please make a PR to MbedOS at the same time.

cmonr, 2018-10-11:
Since it looks like there's a back and forth as to who should open the PR, I'm gonna toss the responsability to @SeppoTakalo IPCore team, since we already have internal progress on the issue, via the GitHub links above.

TuomoHautamaki, 2019-01-15:
Fixed in #9307

closed 2019-01-15 by ciarmcom