ARMmbed/mbed-os #8308: ATHandler::consume_to_tag method doesn't detect tags in some cases

vznncv, 2018-10-02:
### Description
I'm trying to make a driver for a sim5320 using cellular framework, but `ATHandler` doesn't parse some responses correctly. After some investigation, I found that problem is caused by `ATHandler::consume_to_tag` method.

If sequence from buffer contains tag and a symbol before tag coincides with zero symbol of the tag, then the tag won't be detected. For example, if it consumes to a tag `"\r\n"` in a sequence `"\r\r\nOK"`, the tag won't be found.

Problem explanation. The method from the current master branch (SHA: 8b62315)  is the following:
```
bool ATHandler::consume_to_tag(const char *tag, bool consume_tag)
{
    size_t match_pos = 0;

    while (true) {
        int c = get_char();
        if (c == -1) {
            break;
        } else if (c == tag[match_pos]) {
            match_pos++;
            if (match_pos == strlen(tag)) {
                if (!consume_tag) {
                    _recv_pos -= strlen(tag);
                }
                return true;
            }
        } else if (match_pos) {
            match_pos = 0;
        }
    }
    tr_debug("consume_to_tag not found");
    return false;
}
```

Suppose we have want to consume to a tag  `"\r\n"` in a sequence `"\r\r\nOK"`:
1. iteration 1 of the `while` loop: the `\r` is compared with `\r`, symbols are matched and `match_pos` is increased
2. iteration 2 of the `while` loop: the `\r` is compared with `\n`, symbols aren't matched and `match_pos` is set to zero, **but** the code doesn't check if `\r` is matched to the **zero symbol of the tag** (i.e. `\r`).
3. other iterations: the tag won't be found, as its zero symbol is skipped

SHA of the Mbed OS: 8b623156e5fe8e46b6d9ece9101515aa5d37ab13

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

mirelachirica, 2018-10-09:
Internal issue generated.

mirelachirica, 2018-10-10:
PR: https://github.com/ARMmbed/mbed-os/pull/8350/files

cmonr, 2018-10-29:
Marking as closed since #8350 has been merged.

closed 2018-10-29 by cmonr