ARMmbed/mbed-os #8312: Timer read_ms and read_us should return uint64_t instead of int

Arjun765, 2018-10-03:
### Description

```
int Timer::read_us()
{
    return read_high_resolution_us();
}

int Timer::read_ms()
{
    return read_high_resolution_us() / 1000;
}

us_timestamp_t Timer::read_high_resolution_us()
{
    core_util_critical_section_enter();
    us_timestamp_t time = _time + slicetime();
    core_util_critical_section_exit();
    return time;
}
```

### Issue request type

[ ] Question  
[x ] Enhancement  
[ ] Bug

kjbracey-arm, 2018-10-03:
Those continue to return `int` for backwards compatibility. You can use `read_high_resolution_us()` to get the 64-bit value.

Maybe this needs better documentation. In particular the only thing [this page](https://os.mbed.com/docs/v5.9/reference/timer.html) says about range is that the entire class is 32-bit, which is no longer true.

0xc0170, 2018-10-11:
> @kjbracey-arm Would it make sense to just delete that "Note: Timers are based on 32-bit int microsecond counters" section in the API documentation then?

Please send an update to the docs, and tag us for review

cmonr, 2018-10-29:
Instead of changing the API, docs are being updated here: https://github.com/ARMmbed/mbed-os-5-docs/pull/772

closed 2018-11-14 by AnotherButler