ARMmbed/mbed-os #8314: NRF52832 WDT timer interrupt not being called

RobMeades, 2018-10-03:
### Description
I am setting up and successfully running a watchdog timer on an NRF52832 platform (UBLOX_EVK_NINA_B1).  The watchdog fires as expected.  However, I need to do some things when it fires and so I am configuring the WDT interrupt, which should be called 2 x 32 kHz clock cycles before the watchdog resets the chip.  Unfortunately it is never being called.  I've talked to Nordic who have performed a check based on my sample code (see below) and they cannot reproduce the problem: for them the watchdog interrupt is called as expected.

So there must be something else going on: could SoftDevice (which is present in the build but not being called) or anything else in mbed-os be having an effect?

Here is my sample code which shows the problem:

```
#include "mbed.h"
#include <SerialWireOutput.h>

// An unsigned int in an uninitialised RAM area
#if defined(__CC_ARM) || (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))
#error can't make this work, see issue #8289
#elif defined(__GNUC__)
__attribute__ ((section(".noinit")))
unsigned int gRetained;
#elif defined(__ICCARM__)
unsigned int gRetained @ ".noinit";
#endif

// Hook into the weak function to allow Serial Wire Output (because that's how my board speaks to the world)
namespace mbed {
    FileHandle *mbed_target_override_console(int)
    {
        static SerialWireOutput swo;
        return &swo;
    }
}

// Watchdog interrupt handler
void WDT_IRQHandler(void)
{
    gRetained++;
    NRF_WDT->EVENTS_TIMEOUT = 0;
}

// Entry point
int main()
{
    printf("\nStarting up and setting the watchdog timer to 5 seconds.\n");
    // Setting watchdog to 10 seconds
    // Set timeout value timeout [s] = ( CRV + 1 ) / 32768
    NRF_WDT->CRV = (5 * 32768) - 1;
    NVIC_SetPriority(WDT_IRQn, 7);
    NVIC_ClearPendingIRQ(WDT_IRQn);
    NVIC_EnableIRQ(WDT_IRQn);
    NRF_WDT->INTENSET = 1;
    NRF_WDT->TASKS_START = 1;

    printf("Retained RAM variable is %d.\n", gRetained);
    if (gRetained > 2) {
        printf("Setting retained RAM variable to 0 and resetting...\n");
        gRetained = 0;
        wait_ms(1000);
        NVIC_SystemReset();
    }

    gRetained++;
    printf("Retained RAM variable incremented to %d.\n", gRetained);

    if (gRetained < 2) {
        printf("Resetting...\n");
        wait_ms(1000);
        NVIC_SystemReset();
    } else {
        printf("Now waiting for 10 seconds so that the watchdog goes off, which should increment the retained RAM variable to %d.\n",
               gRetained + 1);
        wait_ms(10000);
    }
 
    printf("Should never get here.\n");
    while(1) {}
}
```

...and here is the output:

```
Starting up and setting the watchdog timer to 5 seconds.
Retained RAM variable is 0.
Retained RAM variable incremented to 1.
Resetting...

Starting up and setting the watchdog timer to 5 seconds.
Retained RAM variable is 1.
Retained RAM variable incremented to 2.
Now waiting for 10 seconds so that the watchdog goes off, which should increment the retained RAM variable to 3.

Starting up and setting the watchdog timer to 5 seconds.
Retained RAM variable is 2.
...
```

Any clues welcomed.

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

RobMeades, 2018-10-03:
Note: there's a chance this has the same root cause as #8289, CC @screamerbg @pan- @marcuschangarm.

marcuschangarm, 2018-10-04:
Looks like you are missing the `NVIC_SetVector` call to register `WDT_IRQHandler` as your ISR.

TacoGrandeTX, 2018-12-21:
@RobMeades - Did installing the vector handler as suggested by @marcuschangarm fix the problem?  If so, can we close this question?