ARMmbed/mbed-os #8321: MUTEX_LOCK_FAILED when attempting to read ADC in ISR in 5.10

Arjun765, 2018-10-04:
### Description
```
static AnalogIn adc(PB_0);
static void onTicker()
{
        adc.read();
}

int main()
{
        Ticker t;
        t.attach(onTicker, 1);

        wait(osWaitForever);
}
```

The above code works on 5.9.7 but crashes with MUTEX_LOCK_FAILED in 5.10

### Issue request type

[ ] Question  
[ ] Enhancement  
[x ] Bug

0xc0170, 2018-10-04:
From AnalogIn class: ` * @note Synchronization level: Thread safe`. The object locks using mutex thus not allowed to be invoked from an interrupt handler. 

It should have reported an error even in 5.9.x

What target are you using?

Arjun765, 2018-10-04:
> From AnalogIn class: `* @note Synchronization level: Thread safe`. The object locks using mutex thus not allowed to be invoked from an interrupt handler.
> 
> It should have reported an error even in 5.9.x

Then what is the recommended way to read ADC every few milliseconds?

> What target are you using?

BLUEPILL_F103C8

EmbedEngineer, 2018-11-20:
@0xc0170 I would be very interested in the answer to that topic too. So what is the recommended way to read an ADC e.g. every millisecond or even at a higher frequency?

0xc0170, 2018-11-20:
https://os.mbed.com/blog/entry/Simplify-your-code-with-mbed-events/ - should answer it

Arjun765, 2018-11-21:
> https://os.mbed.com/blog/entry/Simplify-your-code-with-mbed-events/ - should answer it

Mbed events need lots of flash.

kjbracey-arm, 2018-11-21:
Events will certainly need some, but it's a shared component used by quite a few other bits of middleware.

If your system is really minimal, so doesn't already use them, and you don't want to increase your footprint, an alternative is to inherit from `AnalogIn` to make an `UnlockedAnalogIn`, overriding the `lock` and `unlock` methods to do nothing.

That will effectively restore the behaviour you got in 5.9, where the mutex locks were failing silently in interrupt and not giving you any protection.

It's then up to you to ensure that you aren't accessing the object from multiple threads, or from both thread and interrupt at the same time - HALs are not re-entrant, so doing so would cause undefined behaviour.

closed 2019-01-04 by Arjun765