ARMmbed/mbed-os #8322: Travis: astyle package error

0xc0170, 2018-10-04:
### Description

```
curl -L0 http://mbed-os.s3-eu-west-1.amazonaws.com/builds/deps/astyle_3.1_linux.tar.gz --output astyle.tar.gz; mkdir -p BUILD && tar xf astyle.tar.gz -C BUILD; pushd BUILD/astyle/build/gcc; make; export PATH=$PWD/bin:$PATH; popd;
tar: This does not look like a tar archive
```

We fetch astyle package from s3, is s3 corrupted or what has change within last 24 hours?

@ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-test 

We are now getting errors in Travis, see the report https://travis-ci.org/ARMmbed/mbed-os/jobs/436740272 (one of the first).

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-10-04:
If I run curl to fetch the package, previously was working. I get `The specified key does not exist.`

0xc0170, 2018-10-04:
```
curl -L0 http://mbed-os.s3-eu-west-1.amazonaws.com/builds/deps/astyle_3.1_linux.tar.gz --output astyle.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   306    0   306    0     0    306      0 --:--:-- --:--:-- --:--:--  1949

```

The size information is completely wrong. Seems like the package is invalid

OPpuolitaival, 2018-10-04:
should be fixed now

0xc0170, 2018-10-04:
I can confirm locally, restarting travis jobs now

OPpuolitaival, 2018-10-04:
Quick fix done now.. long term fix will be added later with internal ticket number MBEDOSTEST-234

0xc0170, 2018-10-04:
Travis again green. I'll close this as resolved, the long term fix tracked as separate issue.

closed 2018-10-04 by 0xc0170