ARMmbed/mbed-os #8326: Single automated test script error

ganesh-ramachandran, 2018-10-04:
### Description

**Target:**
Tried with Toshiba's targets.

**Toolchains:**
GCC_ARM - 6.3.1
ARM - 5.06
IAR - 7.8

**Issue:**
While doing single test, getting error `TypeError: coercing to Unicode: need string or buffer, list found` from version 5.10 and master. No error in version 5.9.

**Steps to reproduce:**

1. Download master from https://github.com/ARMmbed/mbed-os
2. Build command

```
python tools/singletest.py --auto --tc ARM,GCC_ARM,IAR -v
```
### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-10-05:
@ARMmbed/mbed-os-tools

cmonr, 2018-10-08:
@ganesh-ramachandran Thanks for the issue.
Would you mind providing the entire output, which should also include the traceback?

ganesh-ramachandran, 2018-10-09:
> Would you mind providing the entire output, which should also include the traceback?

Hi @cmonr, no issue in mbed 5.9.7 and issue occurs from 5.10.0.

**Below is the versions of different tools:**
```
- Windows		: 8.1 pro – 64 bit
- Python		: 2.7.14
- Mbed-cli		: 1.8.2
- Greentea		: 1.4.0
- Mbedhtrun		: 1.4.1
- Pip			: 18.1
- Git Bash(git)		: 2.15.1.windows.2
- Mercurial(hg)		: 4.5
```
Please find the below **traceback report** for mbed 5.10.0, mbed master and mbed 5.9.7.
```
mbed-os-mbed-os-5.10.0
----------------------
Looking for TMPM066 with MBEDLS
Traceback (most recent call last):
  File "tools/singletest.py", line 262, in <module>
    if (singletest_in_cli_mode(single_test)):
  File "E:\mbed_singletest\mbed-os-mbed-os-5.10.0\tools\test_api.py", line 1552, in singletest_in_cli_mode
    test_summary, shuffle_seed, test_summary_ext, test_suite_properties_ext, build_report, build_properties = single_test.execute()
  File "E:\mbed_singletest\mbed-os-mbed-os-5.10.0\tools\test_api.py", line 656, in execute
    self.execute_thread_slice(q, target, toolchains, clean, test_ids, self.build_report, self.build_properties)
  File "E:\mbed_singletest\mbed-os-mbed-os-5.10.0\tools\test_api.py", line 585, in execute_thread_slice
    handle_results = self.handle(test_spec, target, toolchain, test_loops=test_loops)
  File "E:\mbed_singletest\mbed-os-mbed-os-5.10.0\tools\test_api.py", line 1030, in handle
    handle_result = self.handle_mut(mut, data, target_name, toolchain_name, test_loops=test_loops)
  File "E:\mbed_singletest\mbed-os-mbed-os-5.10.0\tools\test_api.py", line 945, in handle_mut
    if not exists(image_path):
  File "C:\yotta\python\lib\genericpath.py", line 26, in exists
    os.stat(path)
TypeError: coercing to Unicode: need string or buffer, list found

mbed-os(master)
---------------
Looking for TMPM066 with MBEDLS
Traceback (most recent call last):
  File "tools/singletest.py", line 262, in <module>
    if (singletest_in_cli_mode(single_test)):
  File "E:\mbed_singletest\mbed-os\tools\test_api.py", line 1552, in singletest_in_cli_mode
    test_summary, shuffle_seed, test_summary_ext, test_suite_properties_ext, build_report, build_properties = single_test.execute()
  File "E:\mbed_singletest\mbed-os\tools\test_api.py", line 656, in execute
    self.execute_thread_slice(q, target, toolchains, clean, test_ids, self.build_report, self.build_properties)
  File "E:\mbed_singletest\mbed-os\tools\test_api.py", line 585, in execute_thread_slice
    handle_results = self.handle(test_spec, target, toolchain, test_loops=test_loops)
  File "E:\mbed_singletest\mbed-os\tools\test_api.py", line 1030, in handle
    handle_result = self.handle_mut(mut, data, target_name, toolchain_name, test_loops=test_loops)
  File "E:\mbed_singletest\mbed-os\tools\test_api.py", line 945, in handle_mut
    if not exists(image_path):
  File "C:\yotta\python\lib\genericpath.py", line 26, in exists
    os.stat(path)
TypeError: coercing to Unicode: need string or buffer, list found

mbed-os-mbed-os-5.9.7
---------------------
Looking for TMPM066 with MBEDLS
Executing 'python host_test.py -d F: -f "E:\mbed_singletest\mbed-os-mbed-os-5.9.7\BUILD\test\TMPM066\ARM\DTCT_1\detect.bin" -p COM6 -t 30 -C 4 --auto -m TMPM066'
Test::Output::Start
MBED: Instrumentation: "COM6" and disk: "F:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
Looking for TMPM066 with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '10'
HOST: Property 'host_test_name' = 'detect_auto'
HOST: Property 'description' = 'Simple detect test'
HOST: Property 'test_id' = 'DTCT_1'
HOST: Start test...
{{start}}
HOST: Detecting target name...
MBED: Target 'TMPM066'
HOST: MUT Target name 'TMPM066', expected 'TMPM066'... [OK]
MBED: Test ID 'DTCT_1'
MBED: UUID '8155c14d-6626-458a-8f30-4a6fc444f40f'

{{success}}
{{end}}
Test::Output::Finish
TargetTest::TMPM066[9904]::ARM::DTCT_1::Simple detect test [OK] in 0.52 of 10 sec
Test summary:
+--------+---------------+-----------+---------+--------------------+--------------------+---------------+-------+
| Result | Target        | Toolchain | Test ID | Test Description   | Elapsed Time (sec) | Timeout (sec) | Loops |
+--------+---------------+-----------+---------+--------------------+--------------------+---------------+-------+
| OK     | TMPM066[9904] | ARM       | DTCT_1  | Simple detect test |        0.52        |       10      |  1/1  |
+--------+---------------+-----------+---------+--------------------+--------------------+---------------+-------+
Result: 1 OK

Completed in 8.64 sec

Build successes:
  * TMPM066::ARM::DTCT_1
  * TMPM066::ARM::MBED
```

ganesh-ramachandran, 2019-07-25:
Closing the issue as this issue is not seen with latest mbed-os.

closed 2019-07-25 by ganesh-ramachandran