ARMmbed/mbed-os #8344: [Mbed 2] Cannot compile for NRF51_MICROBIT & ARM - IndexError: list index out of range

yennster, 2018-10-08:
### Description

- Target: NRF51_MICROBIT
- Tools: Mbed CLI
- Toolchain: ARMC5 (5.06u6)
- Mbed 2 version: 170:e95d10626187

Steps to reproduce: 
1. `mbed import http://os.mbed.com/teams/mbed/code/mbed_blinky/`
1. `mbed compile -t arm -m nrf51_microbit -c -v`

Error output:
```
C:\Users\jenplu01\Desktop\mbed_blinky>mbed compile -t arm -m nrf51_microbit -c -v
[mbed] Working path "C:\Users\jenplu01\Desktop\mbed_blinky" (program)
[mbed] Exec "c:\python27\python.exe -u C:\Users\jenplu01\Desktop\mbed_blinky\.temp\tools\make.py -t arm -m nrf51_microbit --source . --build .\BUILD\NRF51_MICROBIT\ARM -c -v" in "C:\Users\jenplu01\Desktop\mbed_blinky"
Using targets from C:\Users\jenplu01\Desktop\mbed_blinky\mbed\targets.json
Building project mbed_blinky (NRF51_MICROBIT, ARM)
Scan: mbed_blinky
Macros: -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K_BASE -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_NRF51_MICROBIT -D__CMSIS_RTOS -DTARGET_MCU_NRF51_16K -DTARGET_NRF_LFCLK_RC -DTARGET_MCU_NRF51_S110 -DTARGET_CORTEX_M -DARM_MATH_CM0 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTOOLCHAIN_ARM_MICRO -DTARGET_RELEASE -DTARGET_NORDIC -DTARGET_MCU_NORDIC_16K -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_MCU_NRF51822 -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1539028352.24 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0
Compile [100.0%]: main.cpp
Compile: C:\Program Files (x86)\ARM_Compiler_5.06u6\bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DMBED_RTOS_SINGLE_THREAD -D__MICROLIB --library_type=microlib --cpu=Cortex-M0 --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti --no_vla -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K_BASE -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_NRF51_MICROBIT -D__CMSIS_RTOS -DTARGET_MCU_NRF51_16K -DTARGET_NRF_LFCLK_RC -DTARGET_MCU_NRF51_S110 -DTARGET_CORTEX_M -DARM_MATH_CM0 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTOOLCHAIN_ARM_MICRO -DTARGET_RELEASE -DTARGET_NORDIC -DTARGET_MCU_NORDIC_16K -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_MCU_NRF51822 -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1539028352.24 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0 --preinclude=.\BUILD\NRF51_MICROBIT\ARM\mbed_config.h --via .\BUILD\NRF51_MICROBIT\ARM\.includes_183a12841efd6c3408f5de2091db44ac.txt --depend BUILD\NRF51_MICROBIT\ARM\main.d -o BUILD\NRF51_MICROBIT\ARM\main.o .\main.cpp
[DEBUG] Return: 0
Traceback (most recent call last):
  File "C:\Users\jenplu01\Desktop\mbed_blinky\.temp\tools\make.py", line 293, in <module>
    ignore=options.ignore
  File "C:\Users\jenplu01\Desktop\mbed_blinky\.temp\tools\build_api.py", line 600, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "C:\Users\jenplu01\Desktop\mbed_blinky\.temp\tools\toolchains\__init__.py", line 624, in link_program
    if path.endswith(self.LINKER_EXT)][-1]
IndexError: list index out of range
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\jenplu01\Desktop\mbed_blinky"
       Command: "c:\python27\python.exe -u C:\Users\jenplu01\Desktop\mbed_blinky\.temp\tools\make.py -t arm -m nrf51_microbit --source . --build .\BUILD\NRF51_MICROBIT\ARM -c -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

yennster, 2018-10-08:
@theotherjimmy This error still occurs when building with Mbed 2 source (https://os.mbed.com/users/mbed_official/code/mbed/ - downloaded as a .zip and extracted into mbed_blinky project) and updating the `arm.py` script with the changes in #8305

0xc0170, 2018-10-10:
> @theotherjimmy With your changes in #8250, it says missing linker script

It should not miss the linker script (microbit target inherits from `MCU_NRF51` that has it defined `MCU_NRF51822`  - this is where the linker script is MCU_NRF51822/device

yennster, 2018-10-10:
@0xc0170 I implemented the changes in #8250 and #8305 in the `.temp\tools` directory in my `mbed_blinky` project (that uses Mbed OS 2) and still receive the following error output:

```
C:\Users\jenplu01\Repos\mbed_blinky>mbed compile -t arm -m nrf51_microbit -c
Using targets from C:\Users\jenplu01\Repos\mbed_blinky\mbed\e95d10626187\targets.json
Building project mbed_blinky (NRF51_MICROBIT, ARM)
Scan: mbed_blinky

Could not compile for NRF51_MICROBIT: No Linker Script found!
```

jpiat, 2018-10-11:
same problem here for a different project. I made the changes mentioned but still cannot compile with error

`Could not compile for NRF51_MICROBIT: No Linker Script found!`

yennster, 2018-10-11:
@jpiat Looks like there is missing files for the micro:bit and this likely won't be fully fixed until the next Mbed 2 release (according to @bridadan & @theotherjimmy)

yennster, 2018-10-23:
@theotherjimmy What is the status on the fix for the micro:bit?

mark-edgeworth, 2019-09-11:
I get 'unable to clone repository' using the repo in the original post.

mark-edgeworth, 2019-09-11:
Following the instructions above I get a missing 'mbed.h' file.

0xc0170, 2021-06-21:
Mbed 2 was deprecated. I'll close this as won't fix.

closed 2021-06-21 by 0xc0170