ARMmbed/mbed-os #8351: Callback uses ROM even when not used

janjongboom, 2018-10-09:
### Description

In mbed-os-example-lorawan there are two radio drivers included from https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers. Only one is selected then through a macro (SX1276), so I'd expect the other one to be linked out. However... 

```
| mbed-lora-radio-drv/SX1272/SX1272_LoRaRadio.o                                                               |   455 |     0 |    0 |
| mbed-lora-radio-drv/SX1276/SX1276_LoRaRadio.o                                                               |  8139 |     0 |    0 |
```

The SX1272 uses 455 bytes of ROM (in `.text`). According to @hasnainvirk this is due to `Callback`'s using ROM even when the object that encapsulates the declaration is linked out. Something similar I saw in https://github.com/ARMmbed/mbed-os/issues/7593.

Can this be fixed? @geky 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

pan-, 2018-10-18:
@janjongboom Remember that template classes and template functions gets instantiated whenever required during the compilation of a C++. So if file _A.cpp_ and file _B.cpp_ use `Callback<void()>`, the code of `Callback<void()>` is present in both _A.o_ and _B.o_ . 

During link, a single definition of each function is selected and kept in the executable.  If the public interface of _A.cpp_ is not used but the linker choose to keep its instantiation of `Callback<void()>` then it will appear like the linker does not eliminate code from _A.cpp_ which is not true; it is just an harmless side effect of templates instantiation.

kjbracey-arm, 2018-10-18:
This is annoying when trying to analyse memory use, but it is effectively harmless.

Although, interestingly, once when I was trying to determine the real user of such templates, I tried just deleting the feature directories that the instantiations were popping up in to try to get them to show against the real code.

And deleting that unused code actually **increased** image size - some alignment issue meant that we ended up with more padding when templates were instantiated elsewhere. Weird.

kjbracey-arm, 2018-10-18:
Anyway, you can test Hasnain's assertion that it's just templates by experimentally deleting (or mbedignoring) mbed-lora-radio-drv/SX1272. You should see +455 or so (depending on padding) get added to another file.

I wonder if there's anything that can be done straightforwardly to distinguish in the memory analysis tool - are the templates placed in a different "common" area that can be distinguished from the "normal" code, to appear as a separate column or total?

pan-, 2018-10-18:
@kjbracey-arm When we enable C++11 or later we could forward common declarations of Callbacks like `void()` and explicitly instantiate them in a dedicated C++ file.

geky, 2018-10-18:
That's interesting! You learn something new every day.

closed 2018-10-29 by ciarmcom