ARMmbed/mbed-os #8353: Unittest folder location

EmbedEngineer, 2018-10-09:
### Description
Hi,

I would like to write unit tests using the newly introduced gtest framework in mbed os. The problem is that the UNITTESTS folder is still being built in the normal build process an therefore I get the error that `fatal error: gtest/gtest.h: No such file or directory #include "gtest/gtest.h"`. 
So do I need a special location for that folder?
My current folder structure is:
´´´
project
+---mbed-os
|   +---...
|   +---UNITTESTS
|   |   +---...
+---platform
|   +---common
|   +---TESTS
|   |   +---common
|   |   |   \---vector
|   +---UNITTESTS
|   |   \---common
|   |       \---vector
´´´
The class under test is ´project/platform/common/Vector.h´ and for the greentea tests this works fine. As soon as I add the ´project/platform/UNITTESTS/common/test_Vector.cpp´ (as described in the documentation https://os.mbed.com/docs/v5.10/tools/unit-testing.html) I get the above mentioned error when building normally with ´mbed compile´ command (I also added the unittest.cmake file in the same folder).

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

yennster, 2018-10-09:
@EmbedEngineer Try changing your folder structure to the following:

```
mbed-os
+---UNITTESTS
|   +---platform
|   |   +---Vector
|   |   |   +---test_Vector.cpp
|   |   |   +---unittest.cmake
```
Make sure your `unittest.cmake` file looks similar to the following:

```

####################
# UNIT TESTS
####################

set(unittest-sources
)

set(unittest-test-sources
    platform/Vector/test_Vector.cpp
)
```

Also, make a `build` directory in `UNITTESTS`

Then, from `mbed-os/UNITTESTS/build` run `cmake ..`

EmbedEngineer, 2018-10-10:
@yennster Thank you for the fast reply. It is working now.
I have one further question. Is there any plan to have the unit tests working like the greentea tests so that I can have the UNITTESTS folder like the TESTS folder outside of the mbed-os folder?

EmbedEngineer, 2018-10-10:
@yennster Now I stumbled upon the problem that I want to configure my SUT with the mbed_app.json. So i start `mbed test --unittest --app-config <path to my mbed_app.json>` and I would like that the config respectively the defines in it are used. Is this possible or are there any plans to implement this?

yennster, 2018-10-10:
@SeppoTakalo @OPpuolitaival Do you have any insight to add on @EmbedEngineer's questions regarding the new unit testing system?

SeppoTakalo, 2018-10-11:
Hi @EmbedEngineer 

Unit testing uses CMake as its build engine to separate itself from all Mbed OS dependencies. The whole idea was to allow us to test C++ classes isolated from Mbed.

The consequence of this is that build system does not parse any JSON configuration files, or pretty much any of the parameters used by other `mbed ...` commands.

Basically the `mbed test --unittest` is just a wrapper to run `./UNITTEST/mbed_unittest.py`

SeppoTakalo, 2018-10-11:
The idea was discussed and recorded in the backlog, but there is no timeline currently when the test tools team would implement this.

closed 2019-01-18 by TacoGrandeTX

boraozgen, 2020-10-09:
> Is there any plan to have the unit tests working like the greentea tests in a way that the UNITTESTS folders like the TESTS folders can be outside of the mbed-os folder?

Is this still not possible?