ARMmbed/mbed-os #8368: tests-mbed_hal-rtc_time  test fails REALTEK_RTL8195AM only for ARM toolchain

studavekar, 2018-10-10:
### Description

tests-mbed_hal-rtc_time test fails on REALTEK_RTL8195AM

**Target**
EV_COG_AD3029LZ

**Toolchain:**
ARM

**mbed-os sha:**

master

**Steps to reproduce**

Use command run :  `mbed test -m REALTEK_RTL8195AM -t ARM  -n tests-mbed_hal-rtc_tim -v`

Failure 

```
[1539211009.62][CONN][INF] Reset the part and send in new preamble...
[1539211009.62][GLRM][INF] remote resources reset...
[1539211013.94][CONN][INF] resending new preamble '3e65fa3a-036d-47f2-a2f6-61fd73368aac' after 5.00 sec
[1539211013.94][GLRM][TXD] {{__sync;3e65fa3a-036d-47f2-a2f6-61fd73368aac}}
[1539211013.95][urllib3.connectionpool]Starting new HTTP connection (1): iron-gatekeeper:8000
[1539211015.12][urllib3.connectionpool]http://iron-gatekeeper:8000 "PUT /resource/460000000989d8d700000000000000000000000097969902/disconnect HTTP/1.1" 200 2
[1539211015.12][urllib3.connectionpool]Starting new HTTP connection (1): iron-gatekeeper:8000
[1539211020.40][urllib3.connectionpool]http://iron-gatekeeper:8000 "PUT /resource/460000000989d8d700000000000000000000000097969902/release HTTP/1.1" 200 66
[1539211020.40][HTST][ERR] None
[1539211020.40][HTST][WRN] stopped to consume events due to __notify_sync_failed event
[1539211020.40][HTST][INF] __exit_event_queue received
[1539211020.40][HTST][INF] test suite run finished after 97.68 sec...
[1539211020.40][HTST][INF] CONN exited with code: 0
[1539211020.40][HTST][INF] No events in queue
[1539211020.40][HTST][INF] host test result() call skipped, received: sync_failed
[1539211020.40][HTST][WRN] missing __exit event from DUT
[1539211020.40][HTST][INF] calling blocking teardown()
[1539211020.40][HTST][INF] teardown() finished
[1539211020.40][HTST][INF] {{result;sync_failed}}
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', 'REALTEK_RTL8195AM', '-p', 'DUMMY:9600', '-f', u'"BUILD/tests/REALTEK_RTL8195AM/ARM/TESTS/mbed_hal/rtc_time/rtc_time.bin"', '-e', u'"TESTS/host_tests"', '--grm', 'raas_client:iron-gatekeeper:8000', '-C', '4', '--sync', '5', '-P', '180'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'SYNC_FAILED'
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-mbed_hal-rtc_time
        test case: tests-mbed_hal-rtc_time
mbedgt: test on hardware with target id: DUMMY
mbedgt: test suite 'tests-mbed_hal-rtc_time' ......................................................... SYNC_FAILED in 98.04 sec
        test case: 'tests-mbed_hal-rtc_time' ......................................................... ERROR in 98.04 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3358972372
mbedgt: test suite report:
+-----------------------+-------------------+-------------------------+-------------+--------------------+-------------+
| target                | platform_name     | test suite              | result      | elapsed_time (sec) | copy_method |
+-----------------------+-------------------+-------------------------+-------------+--------------------+-------------+
| REALTEK_RTL8195AM-ARM | REALTEK_RTL8195AM | tests-mbed_hal-rtc_time | SYNC_FAILED | 98.04              | default     |
+-----------------------+-------------------+-------------------------+-------------+--------------------+-------------+
mbedgt: test suite results: 1 SYNC_FAILED
mbedgt: test case report:
+-----------------------+-------------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
| target                | platform_name     | test suite              | test case               | passed | failed | result | elapsed_time (sec) |
+-----------------------+-------------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
| REALTEK_RTL8195AM-ARM | REALTEK_RTL8195AM | tests-mbed_hal-rtc_time | tests-mbed_hal-rtc_time | 0      | 1      | ERROR  | 98.04              |
+-----------------------+-------------------+-------------------------+-------------------------+--------+--------+--------+--------------------+

```

[ ] Question  
[ ] Enhancement  
[X ] Bug  

cc : @0xc0170

mprse, 2018-10-11:
I'm looking into it.

mprse, 2018-10-11:
REALTEK_RTL8195AM should be added to the list of impacted targets.

mprse, 2018-10-11:
Fix for this issue can be found here:
PR https://github.com/ARMmbed/mbed-os/pull/8377

studavekar, 2018-10-12:
@mprse have updated the description.

M-ichae-l, 2018-10-23:
I believe that PR #8377 has solved this issue. We can close this issue.

studavekar, 2018-10-23:
Closing it tests are passing thanks @mprse 

```
1540313516.21][CONN][RXD] on: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)
[1540313516.21][CONN][RXD]
=========================================================
Check boot type form eFuse
SPI Initial.21][CONN][RXD]
Image1 length: 0x4398, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1
SDR Controller InitN][RXD]
mbedmbedmbedmbedmbedmbedmbedmbed
[1540313517.32][CONN][RXD] >>> Running 6 test cases...
[1540313517.32][HTST][INF] sync KV found, uuid=b3a019f1-9400-4aa9-8839-6b4b0bde9de6, timestamp=1540313517.323582
[1540313517.32][CONN][INF] found SYNC in stream: {{__sync;b3a019f1-9400-4aa9-8839-6b4b0bde9de6}} it is #0 sent, queued...
[1540313517.32][HTST][INF] DUT greentea-client version: 1.3.0
[1540313517.32][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1540313517.32][HTST][INF] setting timeout to: 20 sec
[1540313517.32][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1540313517.32][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1540313517.32][HTST][INF] host test setup() call...
[1540313517.32][HTST][INF] CALLBACKs updated
[1540313517.32][HTST][INF] host test detected: default_auto
[1540313517.32][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1540313517.32][CONN][INF] found KV pair in stream: {{__testcase_count;6}}, queued...
[1540313517.33][CONN][INF] found KV pair in stream: {{__testcase_name;test is leap year - RTC leap years full support}}, queued...
[1540313517.33][CONN][INF] found KV pair in stream: {{__testcase_name;test is leap year - RTC leap years partial support}}, queued...
[1540313517.42][CONN][RXD]
[1540313517.42][CONN][RXD] >>> Running case #1: 'test is leap year - RTC leap years full support'...
[1540313517.42][CONN][INF] found KV pair in stream: {{__testcase_name;test make time boundary values - RTC leap years full support}}, queued...
[1540313517.42][CONN][INF] found KV pair in stream: {{__testcase_name;test make time boundary values - RTC leap years partial support}}, queued...
[1540313517.42][CONN][INF] found KV pair in stream: {{__testcase_name;test make time - invalid param}}, queued...
[1540313517.42][CONN][INF] found KV pair in stream: {{__testcase_name;test local time - invalid param}}, queued...
[1540313517.42][CONN][INF] found KV pair in stream: {{__testcase_start;test is leap year - RTC leap years full support}}, queued...
[1540313517.52][CONN][RXD] >>> 'test is leap year - RTC leap years full support': 1 passed, 0 failed
[1540313517.52][CONN][RXD]
[1540313517.52][CONN][RXD] >>> Running case #2: 'test is leap year - RTC leap years partial support'...
[1540313517.52][CONN][INF] found KV pair in stream: {{__testcase_finish;test is leap year - RTC leap years full support;1;0}}, queued...
[1540313517.52][CONN][INF] found KV pair in stream: {{__testcase_start;test is leap year - RTC leap years partial support}}, queued...
[1540313517.52][CONN][INF] found KV pair in stream: {{__testcase_finish;test is leap year - RTC leap years partial support;1;0}}, queued...
[1540313517.62][CONN][RXD] >>> 'test is leap year - RTC leap years partial support': 1 passed, 0 failed
[1540313517.62][CONN][RXD]
[1540313517.62][CONN][RXD] >>> Running case #3: 'test make time boundary values - RTC leap years full support'...
[1540313517.62][CONN][INF] found KV pair in stream: {{__testcase_start;test make time boundary values - RTC leap years full support}}, queued...
[1540313517.62][CONN][INF] found KV pair in stream: {{__testcase_finish;test make time boundary values - RTC leap years full support;1;0}}, queued...
[1540313517.73][CONN][RXD] >>> 'test make time boundary values - RTC leap years full support': 1 passed, 0 failed
[1540313517.73][CONN][RXD]
[1540313517.73][CONN][RXD] >>> Running case #4: 'test make time boundary values - RTC leap years partial support'...
[1540313517.73][CONN][INF] found KV pair in stream: {{__testcase_start;test make time boundary values - RTC leap years partial support}}, queued...
[1540313517.73][CONN][INF] found KV pair in stream: {{__testcase_finish;test make time boundary values - RTC leap years partial support;1;0}}, queued...
[1540313517.83][CONN][RXD] >>> 'test make time boundary values - RTC leap years partial support': 1 passed, 0 failed
[1540313517.83][CONN][RXD]
[1540313517.83][CONN][RXD] >>> Running case #5: 'test make time - invalid param'...
[1540313517.83][CONN][RXD] >>> 'test make time - invalid param': 1 passed, 0 failed
[1540313517.83][CONN][RXD]
[1540313517.83][CONN][RXD] >>> Running case #6: 'test local time - invalid param'...
[1540313517.83][CONN][INF] found KV pair in stream: {{__testcase_start;test make time - invalid param}}, queued...
[1540313517.83][CONN][INF] found KV pair in stream: {{__testcase_finish;test make time - invalid param;1;0}}, queued...
[1540313517.83][CONN][INF] found KV pair in stream: {{__testcase_start;test local time - invalid param}}, queued...
[1540313517.94][CONN][RXD] >>> 'test local time - invalid param': 1 passed, 0 failed
[1540313517.94][CONN][RXD]
[1540313517.94][CONN][RXD] >>> Test cases: 6 passed, 0 failed
[1540313517.94][CONN][RXD] {{__cpu_info        up time;0}}
[1540313517.94][CONN][RXD] {{__cpu_info     sleep time;0}}
[1540313517.94][CONN][INF] found KV pair in stream: {{__testcase_finish;test local time - invalid param;1;0}}, queued...
[... cut ...]
| REALTEK_RTL8195AM-ARM | REALTEK_RTL8195AM | tests-mbed_hal-rtc_time | test make time boundary values - RTC leap years full support    | 1      | 0      | OK     | 0.0                |
| REALTEK_RTL8195AM-ARM | REALTEK_RTL8195AM | tests-mbed_hal-rtc_time | test make time boundary values - RTC leap years partial support | 1      | 0      | OK     | 0.0                |
+-----------------------+-------------------+-------------------------+-----------------------------------------------------------------+--------+--------+--------+--------------------+
```

closed 2018-10-23 by studavekar