ARMmbed/mbed-os #8372: STM32: Build Fails If Serial Device is Removed

mattbrown015, 2018-10-11:
### Description
Removing the serial device causes the build to fail because of changes made in [STM32: check for UART ongoing transfers before entering deepsleep #7717](https://github.com/ARMmbed/mbed-os/pull/7717).

This feels like a minor issue as I guess it is pretty unusual for anyone to remove the serial device.  _But_, it is supported and the problem occurs because `serial_api.c` is guarded by `#if DEVICE_SERIAL`.

Either the use of `serial_is_tx_ongoing` should be guarded or a stub provided.

### Issue request type
[ ] Question
[ ] Enhancement
[X] Bug

bcostm, 2018-10-12:
Hi
Can you please check if PR #8403 fixes this issue ?

mattbrown015, 2018-10-12:
Hi @bcostm,

I tried your change from #8403, it worked for me. :-)

I wondered if you needed to repeat the comment, including its 2 spelling mistakes, though??  (I'm sure [Skitt's Law](https://knowyourmeme.com/memes/skitt-s-law) applies here!)

bcostm, 2018-10-13:
OK thanks for confirming. I was thinking the same thing to copy or not the comment... ;-) I'll remove it and fix the typo.