ARMmbed/mbed-os #8373: STM32L4: Understanding Deepsleep Power Consumption

mattbrown015, 2018-10-11:
### Description
I appreciate this is an impossible question, and possibly not even appropriate for this forum, but I would appreciate any thoughts. :-)

We've got a custom board with a [STM32L442KC](https://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32-ultra-low-power-mcus/stm32l4-series/stm32l4x2/stm32l442kc.html) and a [Semtech SX1276](https://www.semtech.com/products/wireless-rf/lora-transceivers/SX1276).  In theory I can measure the current on the STM32 and SX1276 power domains separately.

Table 33 of the STM32L442KC [datasheet](https://www.st.com/resource/en/datasheet/stm32l442kc.pdf) suggests current consumption in Stop 2 mode with Vdd at 3.6 V is **1.86** uA.  Let's round this up and say a _few uA_.

When mbed-os selects deepsleep I measure over 82 uA on the STM32 power domain which in mind is 10x too much.  If I reduce Vdd to 1.8 V the current drops to around 22 uA which surprised me as I don't think this big reduction matches what Table 33 says.

I've been through the GPIO configuration and satisfied myself that the pull-up/down configuration is appropriate and shouldn't be causing leakage; I could be wrong.  There are a couple of other components but we've tried to lift their legs.

mbed-os is configured for _tickless_ and `LPTICKER` is in RTC mode.

Thanks in advance!
### Issue request type
[X] Question
[ ] Enhancement
[ ] Bug

mattbrown015, 2018-10-11:
I'm becoming more sure that the current consumption is related to GPIO configuration but it seems to be _complicated_.

If I create a simple RTOS app the current consumption is circa 80 uA:

    #include <mbed.h>

    int main() {
        puts("current-consumption #610e35d " __TIME__ );

        while (1) {
            putchar('.');
            fflush(stdout);

            ThisThread::sleep_for(60 * 1000);
        }
    }

Enable _tickless_ and `LPTICKER` in RTC mode because it is a STM32L4:

    {
        "macros": ["MBED_TICKLESS"],
        "target_overrides": {
            "*": {
                "platform.stdio-baud-rate": "115200"
                , "target.lpticker_lptim": 0
                , "target.lpticker_delay_ticks": 0
            }
    }

Overriding the default configuration of `PA_15` get the current consumption down to 4 uA.  I did this by adding `InterruptIn _dio4_ctl(PA_15);` because this is similar to what the SX1276 radio driver does.

By default `PA_15` is configured as an input with a pull-down and this must be fighting the SX1276 that it is connected to on our board.

_But_, then I found adding `DigitalInOut _reset_ctl(PA_4);` makes the current consumption rise back up 77 uA.  This is similar to what the SX1276 radio driver does.

I shall keep looking at the GPIO configuration.

mattbrown015, 2018-10-11:
Just realised I was getting my pins muddled in the example above `PA_4` is actually the SPI NSS and used as an output by the SX1276 radio driver.

I've moved on to creating a SPI object which causes the current to increase.

LMESTM, 2018-10-11:
For most drivers, the SPI object usually does not include the NSS which is implemented as a GPIO - isn't that your case ?

mattbrown015, 2018-10-11:
Yes, that is the case.

I think I've found an issue though!

Adding a pull-up to MISO reduces the current consumption.

Having a pull-up/pull-down on MISO seems to make sense to me and there are 2 reasons why this wasn't the case:

- The pin map in provided in `TARGET_NUCLEO_L432KC` says `GPIO_NOPULL`
- `pinmap_pinout` in mbed_pinmap_common.c appears to overrule the pinmap and select no pull

From mbed_pinmap_common.c:

    void pinmap_pinout(PinName pin, const PinMap *map)
    {
        if (pin == NC) {
            return;
        }

        while (map->pin != NC) {
            if (map->pin == pin) {
                pin_function(pin, map->function);

                pin_mode(pin, PullNone);
                return;
            }
            map++;
        }
        MBED_ERROR1(MBED_MAKE_ERROR(MBED_MODULE_PLATFORM, MBED_ERROR_CODE_PINMAP_INVALID), "could not pinout", pin);
    }

I changed the pinmap to say `GPIO_PULLUP` and commented `pin_mode(pin, PullNone);`.

Now my deepsleep current consumption is 14 uA. :-)

I start a new issue to discuss `pinmap_pinout`.

LMESTM, 2018-10-11:
whether a pullup / pulldown shuold be applied rather depends on the application and the board (there could be external pull-up). 

Can't you specify the pull-up in your driver, where the NSS pin is declared like
  nss.mode(PullUp);

mattbrown015, 2018-10-11:
Oh, we're getting confused between NSS and MISO.

I was wrong about NSS causing me a problem.

MISO is causing a problem and the MISO pin is configured in the SPI driver according to the pin map.  Well, at least it should be but, AFAICT, isn't.

I understand that what's in `TARGET_NUCLEO_L432KC` might need to be generic hence no pull-ups.  I had just copied it without thinking sometime ago when first getting this all working.